Screenly/Anthias #1765: Images below display resolution are displayed natively and not scaled

MisterAnderson91, 2023-03-03:
Is it possible to have image assets automatically scaled to "Fit" the screen? I.e. scale them up until they fill the screen either horizontally or vertically but don't get cut-off.

At the moment they are just showing at whatever resolution they are surrounded by black, if they are not the native screen res. However larger images are reduced in size automatically. I think it would make sense to scale in both directions.

vpetersson, 2023-03-06:
This is by design. Anthias never scales up images. 

Perhaps we'll add an 'advanced' feature for this one day, but it's not high on the priority list.

closed 2023-03-06 by vpetersson

MisterAnderson91, 2023-03-07:
It seems like an odd design choice for digital signage to leave images too small and bordered for the display they're plugged into. Is there a reason for it?

vpetersson, 2023-03-07:
Scaling up images won't provide a good result. It's there to prevent users from shooting themselves in the foot by quickly providing feedback that the image has too low resolution. Perhaps we should also add warnings in the interface for this, but that's a problem for another day.

MisterAnderson91, 2023-03-07:
I appreciate the concern, but in our case we're using it for a lot of screenshots of timetables and things that just can't be sourced higher res, so we have to perform annoying resize operations to every image. This is non-profit scenario where it's a daily information screen hung on a wall.

It would be great to get an option to just scale them up automatically since it's clear that Anthias could be doing it, but I understand it would have to wait until the project is fully resurrected and working properly.