Screenly/Anthias #1774: Unable to get to web interface nor SSH

pepeuDotCom, 2023-03-08:
I have sort of the same issue as MrAndersson91 at https://github.com/Screenly/Anthias/issues/1773. It's displaying the demo image I put up yesterday when I reinstalled, but today I can't reach it. SSH gets timed out.

nicomiguelino, 2023-03-08:
@pepeuDotCom would you be
 able to provide logs, at least for the server and viewer services/containers?

nicomiguelino, 2023-03-14:
We've already pushed the fix for the broken web UI due to issues caused by migration to Python 3. More specifically, it's a unicode/byte problem.

SSH is not currently possible on Balena images. If  you'd like more flexibility and control, it's recommended to install the RPi OS version.

Closing this issue now...

closed 2023-03-14 by nicomiguelino