Screenly/Anthias #1781: No splash screen, no assests displayed on pi2

qbicdesign, 2023-03-22:
#### Overview of the Issue

No splash screen or assets are displaying, on pi2 after install using pi2 option from pi imager
I had to get ip address from router DHCP list. I was able to access the web ui, and can configure it, add assets, even reboot it etc. 
but all i ever see is the anthias logo

#### Reproduction Steps

Steps to reproduce this issue, eg:

* install clean pi2 image from pi imager
* boot the pi
* anthias logo displays
* splash showing IP address doesn't appear
* assests do not appear
* CTRL + Alt +F2 swithes to black screen, no terminal, and I can't configure via SSH wiythout first enabling it in console...

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 2 Model B Rev 1.1
* Raspberry Pi Network Setup: Ethernet
* Anthias Version: master@137f0a0

Stephim, 2023-04-05:
Same issue on my rpi2

markusguenther, 2023-07-03:
Tried that on a Pi Zero 2 W and had a similar experience. I can add and define new Assets on the web interface, but the HDMI Output shows only the Anthias Logo with a purple background.

salmanfarisvp, 2023-12-11:
@qbicdesign @Stephim @markusguenther Could you please try to install Anthias by running the Script instead of the ready-made image? You can find the step-by-step guide here https://www.hackster.io/Salmanfarisvp/setting-up-anthias-on-raspberry-pi-for-digital-signage-9f6a1b, please check and let me know.

markusguenther, 2023-12-12:
I can give it a try :)
Thanks in advance.