Screenly/Anthias #1782: Asset slideshow gets stuck for a while

MisterAnderson91, 2023-03-26:
Same system as #1773, I came in this morning and checked it. It was showing an image, but that asset didn't change for something like 15 minutes, then it suddenly resumed the 10-second per image slideshow for a while. It's just gotten stuck on another image.

Still getting 504 Gateway Time-out on the web interface so if that comes good I'll double check the display times, but I'm almost certain I didn't change anything off the default 10 second intervals for testing.

ssh access remains working during this so the pi as a whole is still up and running.

nicomiguelino, 2023-03-27:
Would you be able to provide logs from the server, nginx and viewer containers?

MisterAnderson91, 2023-03-28:
```shell
anthias@anthias:~/screenly $ docker compose logs -f anthias-viewer anthias-server anthias-nginx
screenly-anthias-viewer-1  | ./bin/start_viewer.sh: line 39: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
screenly-anthias-viewer-1  | Loading browser...
screenly-anthias-viewer-1  | Generating asset-list...
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/splash-page
screenly-anthias-viewer-1  | ./bin/start_viewer.sh: line 39: /sys/fs/cgroup/memory/memory.swappiness: No such file or directory
screenly-anthias-viewer-1  | Loading browser...
screenly-anthias-viewer-1  | Generating asset-list...
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/splash-page
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
screenly-anthias-viewer-1  | Current url is http://anthias-nginx:80/static/img/loading.png
screenly-anthias-viewer-1  | Playlist is empty. Sleeping for 5 seconds
[... cut ...]
screenly-anthias-server-1  | [2023-03-15 15:09:11 +1000] [20] [INFO] Listening at: http://0.0.0.0:8080 (20)
screenly-anthias-server-1  | [2023-03-15 15:09:11 +1000] [20] [INFO] Using worker: threads
screenly-anthias-server-1  | [2023-03-15 15:09:11 +1000] [28] [INFO] Booting worker with pid: 28
screenly-anthias-server-1  | ConnectionError fetching remote branch availability from GitHub: no data
screenly-anthias-server-1  | Remote Git branch not available
error from daemon in stream: Error grabbing logs: invalid character '\x00' looking for beginning of value
```

MisterAnderson91, 2023-03-28:
I'm not sure what the "playlist is empty" messsage is about, because it's got an asset on screen as we speak (although it's stuck on it right now).

vpetersson, 2023-03-30:
Based on the logs, it *should* be playing the Anthias logo on the screen. Is that not the case?

cfvna271, 2023-04-04:
I'm having this same issue, it's not entirely consistent. Just this morning I came into the office and it's been stuck on one slide for an hour, rebooting the Pi fixes it for me.

MisterAnderson91, 2023-04-18:
> Based on the logs, it _should_ be playing the Anthias logo on the screen. Is that not the case?

Sorry for the delay. No it doesn't show the Anthias logo, it's sitting with one of the uploaded images from the slideshow at that point. I eventually keeps on playing.