Screenly/Anthias #1798: Only 2 dockers running, connection refused on web interface after fresh install

nicosard, 2023-04-16:
Hi,
I have been trying fresh installs,  but after restarting i've no access to the web interface (connection refused), and a ps docker only shows me 2 dockers.

`pi@raspberrypi:~ $ docker ps
CONTAINER ID   IMAGE                                COMMAND                  CREATED              STATUS              PORTS     NAMES
c1851f445715   screenly/anthias-viewer:latest-pi3   "/usr/bin/entry.sh b…"   About a minute ago   Up About a minute             screenly-anthias-viewer-1
ea8e0f496dc2   screenly/anthias-server:latest-pi3   "/usr/bin/entry.sh b…"   About a minute ago   Up About a minute             screenly-anthias-server-1
pi@raspberrypi:~ $ `

Scripts shows me errors, is it related ?  :
` ✔ Network screenly_default                                                                                                  Created                                                                          5.0s 
 ✔ Volume "screenly_resin-data"                                                                                              Created                                                                          0.0s 
 ✔ Volume "screenly_redis-data"                                                                                              Created                                                                          0.0s 
 ⠿ Container screenly-anthias-server-1                                                                                       Starting                                                                         4.1s 
 ⠿ Container screenly-redis-1                                                                                                Starting                                                                         4.1s 
 ✔ Container screenly-anthias-viewer-1                                                                                       Created                                                                          0.6s 
 ✔ Container screenly-anthias-websocket-1                                                                                    Created                                                                          0.5s 
 ✔ Container screenly-anthias-celery-1                                                                                       Created                                                                          0.5s 
 **! anthias-viewer Your kernel does not support memory limit capabilities or the cgroup is not mounted. Limitation discarded.**                                                                                  0.0s 
 ✔ Container screenly-anthias-nginx-1                                                                                        Created                                                                          0.5s 
 ✔ Container screenly-anthias-wifi-connect-1                                                                                 Created                                                                          0.5s 
**Error response from daemon: failed to create endpoint screenly-redis-1 on network screenly_default: failed to add the host (veth1429d90) <=> sandbox (veth62d7b9f) pair interfaces: operation not supported**

...

+ sudo docker system prune -f
Deleted Containers:
8830ad8bba2855b0cb10830d1df3ed852dff25805bdb57102b67f8c3c4da19fe
41d292b341245d8f11014c788fd2e194ed8a8adaa82163f0ff2ebb686859aaa5
e086e651df356ebaa6dcb2493dde737edcee83411168eb32fd20c7c2f019de7c
ebf6eafb72888b5b82a714cf8e9bad3fcdd6f08350c7bab3e64bb3c462ca83fd
5f8a8c53883c0d500efe57262bc41e8a53f8915a7d9ac28bef3e975be7922542
fa7855e4ad8145cd733746551b45e4263d40f2e1a85518e6a3541a9016315ed0
4bd59b6daf757b4dc53f9f429985489e18e499fa60fb2b66e0d4e98c656970c6
`
Installation tested on several µSD cards and several Raspberry Pi (3 and 4). Fresh install of Raspian 11 Bullseye 32bit lite everytime.
It was working fine a few weeks ago.
Thanks for your work.

nicomiguelino, 2023-04-17:
Hello, @nicosard. Thanks for creating an issue! We're currently working on this one right now. There are problems with using the RPi OS Lite installer lately.

Relevant discussions can be seen [here][1].

[1]: https://forums.screenly.io/t/i-simply-cant-get-anthias-to-work/1005/28

nicosard, 2023-04-17:
Thanks for your response.
It seems that the installation works correctly if we don't upgrade before

nicomiguelino, 2023-04-17:
@nicosard, thanks for pointing out that one!
And also, thanks @ealmonte32 for helping out and giving pointers (via discussions outside GitHub).

@ealmonte32 mentioned before (can't recall the PR/issue yet) that the error's gone if we'll chose **_no_** if we're prompted for the system upgrade.

@ealmonte32 & @vpetersson, I was wondering if we could remove prompting for a system upgrade inside `install.sh`. Haven't thought of any reasons (yet) for giving users the choice to opt for an upgrade if it will not work well. I'd love to hear back from you.

vpetersson, 2023-04-17:
> @ealmonte32, @vpetersson, & @rusko124, I was wondering if we could remove prompting for a system upgrade inside install.sh. Haven't thought of any reasons (yet) for giving users the choice to opt for an upgrade if it will not work well. I'd love to hear back from you.

I mean, in theory, that should never break Anthias. If you're doing an initial install, having the system up to date prior is probably not a bad approach.

ealmonte32, 2023-04-17:
@nicosard ,
When you say it works when not choosing to upgrade, did you reboot the Pi after the initial error, and try installation again? Or did you perform a whole fresh install again and this time choose no for upgrade? (when I say fresh install I don't mean restarting the bash install script, I literally mean wipe SD card and reinstall Raspbian OS lite).
When I was troubleshooting this virtual ethernet issue, it would always work after a rpi-update on the fresh installed Raspbian OS, and sometimes just after a reboot, very hard to isolate the actual issue... 
And.. based on what the moby GitHub post said, I could never figure out why after a reboot it would sometimes work without touching the kernel firmware..
...

@nicomiguelino ,
It's weird that updating packages would break the creation of 3 out of 5 containers... but I think if we can prove and isolate that the installation and creation of all containers always works when the updating of the host packages is not done, then for the sake of having less installation errors I would probably also vote for temporarily commenting out that line.. although having a host OS with outdated packages is not the best situation, what if we run it after installation is completed?.. this really all depends on what exactly is causing the container creation errors..

nicomiguelino, 2023-04-18:
Thanks for the idea, @ealmonte32! Sounds like a good workaround. 💡

I've tried doing a proof-of-concept in an isolated environment, and I got similar results (when doing an upgrade prior).

[image]

nicosard, 2023-04-18:
@ealmonte32 ,
From what I've seen on the few installations I've tried :
If I choose "n" at question "Would you like to perform a full system upgrade as well? (y/N)", Anthias installation goes well everytime
If I choose "y" to upgrade, I only have 2 dockers running after install.
If I make an upgrade (apt upgrade) AFTER Anthias install, upgrade fails sometime (on raspberrypi-kernel package if i remember but not sure), but not always ! 
But if upgrade is successful I have all dockers running with the latest system update.
I also tried to upgrade (apt upgrade) and reboot before installing Anthias, but Anthias install also fails.
Each time I start from a fresh RaspiOS 11 lite install (wipe SD card).

nicomiguelino, 2023-04-18:
Hey, thanks @nicosard! I was about to wrap up my day, and then unearthed @vpetersson's [blog post about fixing errors when upgrading][1]. You can give this a try as well. I can try it tomorrow. Thanks.

[1]: https://blog.viktorpetersson.com/2021/05/07/raspberry-pi-fix.html

nicosard, 2023-04-18:
@nicomiguelino thanks for this blog post, i'll test that !

nicomiguelino, 2023-04-18:
@nicomiguelino, did you run `apt upgrade` with `sudo`??
I've tried running `sudo apt -y upgrade`, but was not able to replicate the issue.
How often do you encounter issues when you run `apt upgrade`?

nicosard, 2023-04-19:
@nicomiguelino , Yes I always run apt update && apt upgrade -y with sudo
I would say that I encounter the problem 3 out of 4 times if I upgrade after installing screenly. Never before. 
I haven't tried the blog method yet.

I don't always work with the same storage peripherals (various µSD cards and USB SSD), maybe USB storage causes this problem more often than SD... I would have to try again to better target the problem.

nicomiguelino, 2023-04-26:
Hi, @nicosard. Sorry that it took a while. I created a PR (Screenly/Anthias#1799).
You can take a look at the [README.md][1] as well.

We did the best that we could, and it seems that a reboot is required after a system upgrade.

I'd like to hear back from you, thanks!

[1]: https://github.com/nicomiguelino/Anthias/tree/fix-rpios-installer#installing-on-raspbianraspberry-pi-os

closed 2023-04-27 by nicomiguelino

nicomiguelino, 2023-04-28:
Hey, @nicosard! The PR &ndash; Screenly/Anthias#1799 &ndashl has been merged. How was the experience with running the installer going?