Screenly/Anthias #1810: Anyway to make a slide only appear during specific hours each day?

BeardyMike, 2023-05-04:
#### Overview of the Issue

I have a slide that I want to show every day between 14:20 and 15:50, but I can't find a simple way to do this.

#### Current workaround

Currently I have to load 31 copies of the asset, and set each one to run for a single day between 14:20 and 15:50, and I have to change these dates manually every month.

It'd be great to have an option to have recurring time slot.

#### Environment

* Raspberry Pi Hardware Version: Pi 3B
* Raspberry Pi Network Setup: Wired Ethernet
* Anthias Version: Latest as of 4th May 2023, updated from the players web interface.

vpetersson, 2023-05-10:
This is not a currently supported feature in Anthias. The easiest way to automate will be to use the API.

extremeskillz, 2023-06-06:
The way I got around this with using Anthias as my front end screens and on the backend using a wordpress site with the Foyer plugin and Theater plugin to stream and manage content scheduling. The only thing you do is add the URL asset in wordpress Foyer to the Anthias Asset for streaming/pulling from the wordpress site. My smaller screens in which I don't need this is just using the built assets in Anthias.