Screenly/Anthias #1813: Upgrade failed when using Raspberry Pi 3b with Stretch

Kipjr, 2023-05-10:
#### Overview of the Issue

Unable to upgrade old 2020 Screenly to most recent version. Please implement a warning when OS is < required

#### Reproduction Steps

- Grab raspberry pi 3 with stretch and screenly of 2020 and install it
- Upgrade using `./screen/bin/run_upgrade.sh`

#### Environment

* Raspberry Pi Hardware Version: 3b
* Raspberry Pi Network Setup: Wired
* Screenly Version:  The version I installed on  Sep 26th 2019

#### Solution 
- Found but only after a while of troubleshooting:

1.  sudo nano /etc/apt/sources.list

2. replace `deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi` with 
`deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi`

3. `sudo apt update && sudo apt dist-upgrade` 

4. Retry upgrade of Anthias

vpetersson, 2023-05-10:
Thanks for providing this, @Kipjr. If that worked for you great. For other people landing here, I would recommend reading the [Debian Upgrade](https://wiki.debian.org/DebianUpgrade) in case you are having issues.

closed 2023-05-10 by vpetersson

nicomiguelino, 2023-05-11:
I'll take note of this, @vpetersson & @Kipjr. Thanks! I'll go create an issue (task) for updating the docs.
Will keep you posted. 📻