Screenly/Anthias #1822: Anthias freeze after a few hours

agawley, 2023-07-13:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

After a few hours of running, Anthias freezes and shows the attached error:

`Unhandled prefetch abort: unknown 1 (0x001) at 0x00000066`

#### Reproduction Steps

Steps to reproduce this issue, eg:

Add the following assets with 10s timeout and Forever duration:
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page1
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page2
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page3
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page4
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page5
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page6
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page7
https://lxi2u95r4l.execute-api.eu-west-2.amazonaws.com/dev/page8

#### Environment

* Raspberry Pi Hardware Version: 3 B+ (PiOS Lite latest)
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter) built in WiFi
* Anthias Version: 0.18.6

agawley, 2023-07-13:
[image]

vpetersson, 2023-07-13:
Interesting. By the look of that, that's a kernel error. Do you have any peripheral attached?

agawley, 2023-07-13:
No. Just screen and USB power

vpetersson, 2023-07-13:
What PSU are you using?

Also, just to rule out that it's not a broken SD card (this tend to be the root cause for a bunch of random issues), can you try with another new SD card and see if you are able to reproduce it?

ealmonte32, 2023-07-13:
> No. Just screen and USB power. Are their Anthias logs I can pull to see what is happening on that side?

@agawley , this might not be a log event triggered by Anthias, but if you wanted to troubleshoot:

connect via `ssh` and type: `docker logs screenly-anthias-viewer-1` (if you want to see the viewer container logs)
if you press tab after the commands `docker logs`, it will list for you the available containers, and you can then select which of the containers you want to see the logs for, but again, this error might be a kernel one which is PiOS related, thus you can `ssh` and type `journalctl -r` (i like putting the -r so it shows me in reverse which shows me the latest logs first and then i scroll through to older ones.

ealmonte32, 2023-07-14:
> After a few hours of running, Anthias freezes and shows the attached error:
> 
> `Unhandled prefetch abort: unknown 1 (0x001) at 0x00000066`

I don't give this advise a lot because of issues it can have _(not common to have any bad side-effects)_, but the latest pi kernel firmware after a certain version did supposedly fix this specific kernel error for some users which seems to deal with memory that is supposed to be at that address but cant be found _(based on raspberrypi\kernel readings)_.

Since I have tested this on the same hardware version you have, would you mind running the `sudo rpi-update` command and update to the latest rpi firmware? At the very least you will have updated pi firmware.

agawley, 2023-07-16:
Thanks for the tip. Did this and managed to run for a good amount of time without a crash. Did crash eventually with a different kernel error though so am going to try a new sd card. I'll close this issue as i don't think this is an anthias issue.

closed 2023-07-16 by agawley