Screenly/Anthias #1824: Create tool for installing (trust) self-signed certificates

vpetersson, 2023-08-07:
Replaces #1516

We should have a helper script for installing certificates inside the containers. It would be fairly straight forward and work something like this:

`./bin/add_certificate.sh /path/to/my.crt`

This would in turn install do something like this (run both against the viewer and server container):

* `docker exec -ti container-name mkdir /usr/local/share/ca-certificates/custom`
* `docker cp /path/to/my.crt container-name:/usr/local/share/ca-certificates/custom/`
* `docker exec -ti container-name update-ca-certificates`

Note that the above is completely untested, but should in theory work.

DarthDestroyer, 2023-08-07:
`#!/bin/bash

docker exec -ti screenly-anthias-server-1 mkdir /usr/local/share/ca-certificates/custom
docker exec -ti screenly-anthias-viewer-1 mkdir /usr/local/share/ca-certificates/custom

docker cp /home/pi/import.crt screenly-anthias-server-1:/usr/local/share/ca-certificates/custom/
docker cp /home/pi/import.crt screenly-anthias-viewer-1:/usr/local/share/ca-certificates/custom/

docker exec -ti screenly-anthias-server-1 update-ca-certificates
docker exec -ti screenly-anthias-viewer-1 update-ca-certificates`

From Code-Side, it worked

vpetersson, 2023-08-07:
What happens if you try to `curl` the internal service from within the container? Is the certificate valid then?

vpetersson, 2023-08-08:
Interesting. So when you try adding that as an asset, in what state does it fail?

DarthDestroyer, 2023-08-09:
When i add the URL, everything works flawless! No Error, no interrupt!

But when the Item gets displayed, its written on the screen itself "ERR_CERT_AUTHORITY_INVALID" - The Site cant be reached

DarthDestroyer, 2023-08-09:
*  SSL certificate verify ok.

Also on the Viewer - Sorry! Missed that one - but the Settings and Certs are set to both Containers! So the Viewer curl request worked!

i dont know, what browser is running and why it doesnt use the system certs - but we could have found the issue

DarthDestroyer, 2023-08-09:
Im not that far into QT... But there is a bunch of Results, when you google it!

Ignoring the Cert: https://stackoverflow.com/questions/67459911/qwebengineview-and-ignoring-cert-errors

Adding the Cert: https://doc.qt.io/qt-5/qwebengineclientcertificatestore.html

DatPenguin, 2024-01-26:
I, too, am interested in this feature. Has a way to do this been found yet ?

vpetersson, 2024-01-26:
A (better?) workaround might be to use Tailscale with their [TLS certificates](https://tailscale.com/blog/tls-certs). They're valid certs (issued by Let's Encrypt). Implementing this should be straight forward. You might find [this article](https://blog.viktorpetersson.com/2022/12/23/securing-services-with-tailscale.html) that I wrote some time back helpful for that.

DarthDestroyer, 2024-01-26:
It has bugged a lot, so i throwed everything away and homebrew an alternative...

But yeah, you can get a valid cert and anthias doesnt complain about it!
It isnt a real Solution, but nothing has worked to implement a Company-Cert / Self-Signed Cert

zeeZ, 2024-06-06:
Hello, just randomly chiming in. I don't currently have a device to run as a viewer so I can't comment on that, but I might have some hits on the server side:

python-certifi ships with its own trust store bundle, so updating the default system trust store has effect on the server, even though curl is working.

In code it's explicitly set [here](https://github.com/Screenly/Anthias/blob/v0.18.7/lib/utils.py#L265) with `verify = certifi.where()`. The bundle used is somewhere around here-ish: `/usr/local/lib/python3.7/dist-packages/certifi/cacert.pem` and overriding that allows adding internal HTTPS URLs through the web interface without skipping the asset check.

Additionally, if `settings['verify_ssl']` is `False`, `verify` is set to `True` in the file linked above, which does *not* disable certificate verification, but causes requests to default to the same certifi bundle, but also allows that to be overridden with `REQUESTS_CA_BUNDLE` or `CURL_CA_BUNDLE`. To disable verification, `verify` should be set to `False` here.

 In short, there are currently (1.18.7) two ways to get custom certificates to work **on the server side**:
1. Replace `/usr/local/lib/python3.7/dist-packages/certifi/cacert.pem` (found by executing `python3 -m requests.certs`)
2. In `screenly.conf`, set `verify_ssl = False` in the `[viewer]` section and set the environment variable `REQUESTS_CA_BUNDLE` to a path with your trust bundle. I assume this has unwanted side effects on the viewer though if configuration is shared.

ian-knight-acacia, 2024-07-31:
Hi all,
Ran into this issue myself. For the viewer, found that QWebEngine uses an NSS shared DB rather than the default system certificate store. The db is stored in the resin-data volume at `.pki/nssdb`

To install the cert, use `certutil` from the `libnss3-tools` package:  
`certutil -A -n "My CA Certificate" -t "C,C,C" -i "/path/to/certificate.pem" -d "/path/to/.pki/nssdb"`

Documentation and examples on certutil:  
- <https://wiki.mozilla.org/NSS_Shared_DB_Howto>
- <https://docs.redhat.com/en/documentation/red_hat_certificate_system/9/html/administration_guide_common_criteria_edition/importing_certificate_into_nssdb#importing_certificate_into_nssdb>

vpetersson, 2024-08-14:
This is a really good find. @nicomiguelino can you add this to the general documentation?

nicomiguelino, 2024-08-14:
@vpetersson, I'll test and create a pull request for that one as soon as I can.

PR #2023 (2024-08-14): Creates a script for installing (trusted) self-signed certificates
#### Description

* Fixes #1824
closed 2024-08-14 by nicomiguelino