Screenly/Anthias #1826: Scheduled toggling of screen

colindean, 2023-09-07:
#### Overview of the Issue

I'm trying out Anthias to replace some custom-made signage software built ~7 years ago. It just used Midori to load a browser window with a pre-configured URL specified in a text file on the boot drive, then hid the mouse cursor when idle.

One of my important features is something that turns off the screen at night and turns it back on the morning. My crontab:

```crontab
0       22      *       *       *               vcgencmd display_power 0
0       8       *       *       1,2,3,4,5       vcgencmd display_power 1
```

This has reduced the power usage of our pair of displays by around 25% compared to leaving them on all the time. Since installation in 2017, I reckon this configuration has saved us 

```
2381 days since March 1, 2017
= 340 weeks
10 hrs/day off * 5 days/week
24 hrs/day off * 2 days/week
= 98 hrs/week off
50 W/monitor
2 monitors
= 3,332 kWh
* 15 ¢/kWh
= $499.80
```

~$500 dollars over the last 6.5 years. Not much, but on our budget, that's something.

I'd love to see this feature built into Anthias.

nicomiguelino, 2023-09-07:
Hi there. Thanks for creating this issue! It would be nice to implement this in the future. For now, I don't have the specifics to implement the feature, aside from using a `crontab`.

vpetersson, 2023-09-07:
You could just lift that part and have it run on Anthias to be honest. Just do a raspbian based installation and you should be good to go.

colindean, 2023-09-11:
Thanks for the nudge to go with the RasPiOs/Raspbian installation. I redid that tonight and I've got things setup the way I really want including a new schedule that hits a wider range and a few more days I know the panels can be offline:

```
# standard off
0       21      *       *       *               vcgencmd display_power 0
0       9       *       *       1,2,3,4,5       vcgencmd display_power 1

# holidays on a static date
0       0       25      12      *               vcgencmd display_power 0
0       0       1       1       *               vcgencmd display_power 0
0       0       4       7       *               vcgencmd display_power 0
```

vpetersson, 2023-09-11:
Great! Will close out this issue as it seems to be working.

closed 2023-09-11 by vpetersson