Screenly/Anthias #1829: 4K TV, screen white, cannot display content 3840x2160

frwxworx, 2023-10-12:
#### Overview of the Issue

4K TV (3840x2160 @ 30.00Hz, progressive) - white screen, .jpg .png content displayed for miracle of few ms and abrupted into white, blank screen. 
Even default assets (Weather, Clock widget etc.) flashes quickly and goes full-white screen again.

What I tried so far without positive results, the screen still goes white:

- Added variable RESIN_HOST_CONFIG_hdmi_enable_4kp60 = 1;
- In the docker compose file docker-compose.yml added a link under the viewer service shm_size: '1gb' to enlarge "viewer" container size:
   srly-ose-viewer:
      shm_size: '1gb'

Should I mess with hdmi_group=2 (DMT) and adding custom hdmi_mode and hdmi_cvt=<width> <height> <framerate> <aspect> <margins> <interlace> <rb> resolution or other?

Utterly annoying, we have 4K TVs all around the building. P.S. Legacy Screenly-0SE without Balena OS solution actually works with 4K TV and can display content absolutely fine.

The 4K content playback issue has been mentioned several times before:
https://forums.screenly.io/t/anthias-white-screen-on-certain-kinds-of-websites-screenly-ose-still-working/1096/11
https://github.com/Screenly/Anthias/issues/1498#issuecomment-780993344

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version: Pi 4 Model B Rev 1.2
* Raspberry Pi Network Setup: Ethernet
* Anthias Version: v0.18.6

dr-waterstorm, 2023-10-23:
I can confirm this issue with a new Raspberry Pi 4 Rev 1.5, powered using a PoE-Hat and the latest Anthias.

vpetersson, 2023-10-24:
@traeu The Pi 3 doesn't properly support 4k in the first place. There are some hacks to make it work at very low frame rate, but it's not natively supported.

traeu, 2023-10-24:
Yes, I am aware of that.
But it still works, with low framerate, even if it is not officially supported.
Because I get the same result (blank screen) as Raspi 4, which supports 4K nativly, I hope that if you fix this for Raspi 4, it will also work on older Raspis.
Right now, I don't think that the blank screen problem is a result of "hacking" Raspi 3 into displaying 4K resolution, because the same problem exists on raspi 4.

vpetersson, 2023-10-25:
Not suggesting the root cause is the Pi 3. Just that the Pi 3 is severely under resourced for any kind of 4ik content.

traeu, 2023-10-26:
I guess it really depends on the content you are displaying. I use anthias to show a schedule. It's a very small web-page containing only text and some lines. No images or video content. For me, even 1 Hz framerate would be enough. But there is a visible difference between using 4K and Full HD, the TV screens are big enough that with 4K, fonts and lines look a lot sharper and reading small texts is easier.
So if I need to decide between FPS and resolution, I would choose resolution. No one would notice if that schedule gets refreshed only 1 time per second...

nicomiguelino, 2023-11-02:
@vpetersson, could the root cause be in the Qt/C++ code?

vpetersson, 2023-11-02:
Yes entirely possible. Would be a good test to see if this affects *all* asset types. If it only affects images and web pages, it's likely related to the Qt side of things.

traeu, 2024-01-16:
@vpetersson I had some time to do this test. Its done on Raspi 3b (as you know, no official 4K support).
On Full HD: Webpages load normally, video (h264, 1080p) plays normally.
On 4K: Webpages are only shown for a fraction of a second, then only white screen. video (h264, 1080p) plays normally.

palmarcell, 2024-04-22:
I can sadly still confirm the issue.
I wanted to simply open up a Google Slides document with like 5 slides in it, but since the TV is 4k I can only see a blank white screen.

ckmiecik, 2024-05-10:
I am having the same issue. I am working on a Mac building images in powerpoint and then exporting them. When I save slides from power point as images I have an option to set the resolution in pixels. I have spent hours trying out different resolutions and found that anything over 3603x2027 makes the screen go fully white. Anything I export that is 3603x2027 and lower works but does not fill my screen (since it is a 4k screen). I have also tried exporting images from Adobe indesign but I still get the same white screen. I am using a Raspberry pi4 with a 4K monitor (also same thing on a 4K Visio TV).  All I need is to have images displayed and fill the whole screen (on a 4K screen).  Does anybody have a fix for this?

ckmiecik, 2024-05-10:
When looking at my docker-compose.yml file I see no sim_size.  Is this a line I need to add somewhere?

erobinson1979, 2024-05-14:
I've added it under every container in the file, it's still not working for me.

erobinson1979, 2024-05-14:
> shm_size should be present (at least it was for me). Also setting it to 1gb solved the issue. I found the reference here: #1554

Thanks for the reply and the source info. I was able to edit my file to match that post and remove the extra shm_size entries, but I'm still experiencing the same symptoms as [#1554 ](https://github.com/Screenly/Anthias/issues/1554)

My next step is probably going to be finding a Pi 3b and using the older version that works.

ckmiecik, 2024-05-14:
Thanks for the update.  I also tried editing the file but still the same result.  If I am understanding this correctly it seems like this should be fixed when #1887 is resolved.  The only thing I am worried about, is that it sounds like this is going to fix web pages but I'm hoping it also fixes my issue where I'm trying to just display .jpgs

nicomiguelino, 2024-05-15:
It might take a while for the Docker images to be built. You can update your Anthias instances once the following CI run is finished: https://github.com/Screenly/Anthias/actions/runs/9101858709

closed 2024-05-15 by nicomiguelino

traeu, 2024-05-22:
I can confirm that this fix also works for Raspberry Pi 3B that has no official 4K support.
After updating anthias, I was able to get working 4K output!
Thanks for fixing this.

PR #1894 (2024-05-15): Allows support for 4k display  by increasing shared memory in viewer
#### Overview

* Fixes #1829

#### Installing and testing the changes

```bash
cd && \
echo "export CUSTOM_BRANCH='fix-1829'" > .env && \
bash <(curl -sL https://raw.githubusercontent.com/nicomiguelino/Anthias/custom-install-1/bin/install.sh)
```

#### Checklist

- [x] Fix web asset display in 4k for Raspberry Pi OS.
- [x] Test changes on Bookworm for Pi 3, and Pi 4 devices.
- [x] Fix web asset display in 4k displays for Balena OS.
- [x] Test changes on Bookworm for Pi 2, Pi 3, and Pi 4 devices.