Screenly/Anthias #1833: MP4 video crashes  | Raspberry Pi 4

macxq, 2023-10-19:
#### Overview of the Issue

The MP4 video plays but stops at a completely random moment. This issue always occurs, usually after just a few repetitions, often the first time

`docker restart screenly-anthias-viewer-1` helps to recover

#### Reproduction Steps

Add mp4 video asset

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 2GB
* Raspberry Pi Network Setup: built-in WiFi 
* Anthias Version: master@4170486

Option  | Value
-- | --
| Load Average | 0.17| 
Free Space | 23G
Memory | Total: 1731 / Used: 368 / Free: 694 / Shared: 29 / Buff: 30 / Available: 1265
Uptime  | 0 days and 0.34 hours
Monitor Info |  0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
Display Power (CEC) | None
Raspberry Pi Model | Raspberry Pi 4 Model B Rev 1.1

```
Generating asset-list...
Showing asset cm.mp4 (video)
Current url is null
[ee41ccc0] mmal_codec decoder: VCSM init succeeded: CMA
[ee41ccc0] main decoder error: buffer deadlock prevented
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
```

```
CONTAINER ID   NAME                           CPU %     MEM USAGE / LIMIT     MEM %     NET I/O           BLOCK I/O         PIDS
c088708a50c1   screenly-anthias-viewer-1      8.73%     113.5MiB / 1.353GiB   8.20%     867kB / 1.16MB    0B / 8.19kB       73
a12b407e3b52   screenly-anthias-celery-1      0.20%     89.56MiB / 1.691GiB   5.17%     843kB / 878kB     11.5MB / 4.1kB    12
23368a8558e8   screenly-anthias-nginx-1       0.00%     13.36MiB / 1.691GiB   0.77%     17.4MB / 16.9MB   11MB / 1.72MB     5
a1ff3ece7b6e   screenly-anthias-websocket-1   0.86%     27.34MiB / 1.691GiB   1.58%     514kB / 673kB     19.2MB / 0B       5
ad5831c7ee0e   screenly-redis-1               0.48%     4.039MiB / 1.691GiB   0.23%     924kB / 840kB     3.33MB / 0B       4
d28bdc433f59   screenly-anthias-server-1      0.09%     80.75MiB / 1.691GiB   4.66%     17.7MB / 1.2MB    50.9MB / 16.5MB   10
```

vpetersson, 2023-10-19:
Can you run ffprobe on the video?

macxq, 2023-10-20:
@vpetersson  sure 👍 

```
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from './screenly_assets/05edf45f55ac4a149fb6f61f9b528b4e.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf58.76.100
  Duration: 00:00:15.90, start: 0.000000, bitrate: 1203 kb/s
    Stream #0:0(und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709), 1920x1080, 1199 kb/s, 30 fps, 30 tbr, 15360 tbn, 60 tbc (default)
    Metadata:
      handler_name    : VideoHandler
```

macxq, 2023-10-20:
I get into container and made vlc more verbose: 
```
VLC media player 3.0.17.4 Vetinari (revision 3.0.13-8-g41878ff4f2)
[010bab88] main libvlc debug: VLC media player - 3.0.17.4 Vetinari
[010bab88] main libvlc debug: Copyright © 1996-2022 the VideoLAN team
[010bab88] main libvlc debug: revision 3.0.13-8-g41878ff4f2
[010bab88] main libvlc debug: configured with ./configure  '--build=arm-linux-gnueabihf' '--prefix=/usr' '--includedir=${prefix}/include' '--mandir=${prefix}/share/man' '--infodir=${prefix}/share/info' '--sysconfdir=/etc' '--localstatedir=/var' '--disable-option-checking' '--disable-silent-rules' '--libdir=${prefix}/lib/arm-linux-gnueabihf' '--runstatedir=/run' '--disable-maintainer-mode' '--disable-dependency-tracking' '--disable-debug' '--config-cache' '--disable-update-check' '--enable-fast-install' '--docdir=/usr/share/doc/vlc' '--with-binary-version=3.0.17.4-0+deb10u2+rpt1' '--enable-a52' '--enable-aa' '--enable-aribsub' '--enable-avahi' '--enable-bluray' '--enable-caca' '--enable-chromaprint' '--enable-chromecast' '--enable-dbus' '--enable-dca' '--enable-dvbpsi' '--enable-dvdnav' '--enable-faad' '--enable-flac' '--enable-fluidsynth' '--enable-freetype' '--enable-fribidi' '--enable-gles2' '--enable-gnutls' '--enable-harfbuzz' '--enable-jack' '--enable-kate' '--enable-libass' '--enable-libmpeg2' '--enable-libxml2' '--enable-lirc' '--enable-live555' '--enable-mad' '--enable-matroska' '--enable-mod' '--enable-mpc' '--enable-mpg123' '--enable-mtp' '--enable-ncurses' '--enable-notify' '--enable-ogg' '--enable-opus' '--enable-pulse' '--enable-qt' '--enable-realrtsp' '--enable-samplerate' '--enable-sdl-image' '--enable-sftp' '--enable-shine' '--enable-shout' '--enable-skins2' '--enable-sndio' '--enable-soxr' '--enable-spatialaudio' '--enable-speex' '--enable-svg' '--enable-svgdec' '--enable-taglib' '--enable-theora' '--enable-twolame' '--enable-upnp' '--enable-vnc' '--enable-vorbis' '--enable-x264' '--enable-x265' '--enable-zvbi' '--with-kde-solid=/usr/share/solid/actions/' '--disable-d3d11va' '--disable-decklink' '--disable-directx' '--disable-dsm' '--disable-dxva2' '--disable-fdkaac' '--disable-fluidlite' '--disable-freerdp' '--disable-goom' '--disable-gst-decode' '--disable-libtar' '--disable-macosx' '--disable-macosx-avfoundation' '--disable-macosx-qtkit' '--disable-microdns' '--disable-mfx' '--disable-opencv' '--disable-projectm' '--disable-schroedinger' '--disable-sparkle' '--disable-srt' '--disable-telx' '--disable-vpx' '--disable-vsxu' '--disable-wasapi' '--enable-alsa' '--enable-dc1394' '--enable-dv1394' '--enable-libplacebo' '--enable-linsys' '--enable-nfs' '--enable-omxil' '--enable-udev' '--enable-v4l2' '--enable-wayland' '--enable-vcd' '--enable-smbclient' '--disable-oss' '--disable-crystalhd' '--disable-mmx' '--disable-sse' '--enable-neon' '--enable-mmal' '--enable-mmal-avcodec' '--disable-vdpau' '--disable-libva' '--enable-dav1d' '--disable-aom' '--disable-altivec' 'build_alias=arm-linux-gnueabihf' 'CFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security ' 'LDFLAGS=-Wl,-z,relro -Wl,-z,now -Wl,--as-needed' 'CPPFLAGS=-Wdate-time -D_FORTIFY_SOURCE=2' 'CXXFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security ' 'OBJCFLAGS=-g -O2 -fdebug-prefix-map=/build/vlc-iTPUTR/vlc-3.0.17.4=. -fstack-protector-strong -Wformat -Werror=format-security'
[010bab88] main libvlc debug: searching plug-in modules
[010bab88] main libvlc debug: loading plugins cache file /usr/lib/arm-linux-gnueabihf/vlc/plugins/plugins.dat
[010bab88] main libvlc debug: recursively browsing `/usr/lib/arm-linux-gnueabihf/vlc/plugins'
[010bab88] main libvlc debug: plug-ins loaded: 501 modules
[010bab88] main libvlc debug: opening config file (/data/.config/vlc/vlcrc)
[010bada8] main logger debug: looking for logger module matching "any": 4 candidates
[010bada8] main logger debug: using logger module "console"
[010bab88] main libvlc debug: translation test: code is "C"
[01120380] main keystore debug: looking for keystore module matching "memory": 4 candidates
[01120380] main keystore debug: using keystore module "memory"
[010bab88] main libvlc debug: CPU has capabilities FPU
[01123e08] main playlist debug: creating audio output
[01115468] main audio output debug: looking for audio output module matching "any": 6 candidates
[01115468] vlcpulse audio output debug: using library version 12.2.0
[01115468] vlcpulse audio output debug:  (compiled with version 12.2.0, protocol 32)
Home directory not accessible: Permission denied
[01115468] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01115468] alsa audio output debug: Available ALSA PCM devices:
[01115468] alsa audio output debug: Discard all samples (playback) or generate zero samples (capture) (null)
[01115468] alsa audio output debug: bcm2835 HDMI 1, bcm2835 HDMI 1 Default Audio Device (default:CARD=b1)
[01115468] alsa audio output debug: bcm2835 HDMI 1, bcm2835 HDMI 1 Default Audio Device (sysdefault:CARD=b1)
[01115468] alsa audio output debug: bcm2835 HDMI 1, bcm2835 HDMI 1 Direct sample mixing device (dmix:CARD=b1,DEV=0)
[01115468] alsa audio output debug: bcm2835 HDMI 1, bcm2835 HDMI 1 Direct sample mixing device (dmix:CARD=b1,DEV=1)
[01115468] alsa audio output debug: bcm2835 HDMI 1, bcm2835 HDMI 1 Direct sample snooping device (dsnoop:CARD=b1,DEV=0)
[... cut ...]
[eb802cd0] main video output warning: picture is too late to be displayed (missing 2300 ms)
[f2800668] main input debug: EOF reached
[f2800668] main input debug: waiting decoder fifos to empty
[f2800668] main input debug: waiting decoder fifos to empty
[f2800668] main input debug: waiting decoder fifos to empty
```
And found two links around that but not sure if helpful 
 
https://forums.raspberrypi.com/viewtopic.php?t=329408
https://forum.videolan.org/viewtopic.php?f=13&t=159158&p=524894&hilit=fifos#p524894

vpetersson, 2023-10-23:
@nicomiguelino Please take a look at this when you get a chance.

nicomiguelino, 2023-11-03:
@macxq, I was not able to replicate the issue on my Pi 4 (Model B, with 4GB of RAM). I haven't experienced random stops yet. Would you be able to provide the actual MP4 video that you've used?
I'm not sure if it's because of the video or because of memory constraints.

rapasae, 2023-11-13:
If it can help, I have the exact same problem and it doesn't seem to depend on the video encoding. 
I have tried encoding them with different formats, resolutions, bit rates, etc... In my case I have noticed that this only happens when the videos are played from a remote url, if they are played from the raspberry's SD, it does not crash.

vpetersson, 2023-11-13:
Playing videos from a remote URL is strongly discouraged in general.

nicomiguelino, 2024-06-13:
@macxq, are you still having this kind of issue? Anthias switched back to using omxplayer in the meantime. At the time, I'm not experiencing such issues.

nicomiguelino, 2024-08-07:
Anthias recently swtched back from OMX to VLC with the upgrade of Docker containers from Buster to Bookworm (see #1980).

closed 2024-08-07 by nicomiguelino