Screenly/Anthias #1834: /home/$USER/screenly/bin/upgrade_containers.sh does not exist

ChameleonGeek, 2023-10-21:
#### Overview of the Issue

Installed Raspberry Pi OS lite (bookworm).  After update/upgrade, ran bash <(curl -sL https://install-anthias.srly.io).  Rebooted after the script completed.  /home/$USER/screenly/ exists but is empty.

If I run sudo bash <(curl -sL https://install-anthias.srly.io), the script immediately fails with the error "/dev/fd/63: No such file or directory"

Whether I run with default permissions or after sudo su, the directory is empty.  No errors are shown during the install process.

I am attempting this installation approach because installing Anthias via the Raspberry Pi imager results in an eternal hang at the Anthias splash screen.  The system never displays any assets.

vpetersson, 2023-10-21:
@nicomiguelino can you try to reproduce this?

nicomiguelino, 2023-10-21:
@vpetersson, @ChameleonGeek
I was able to reproduce the issue. Here's a part of what I've got after running the installation script on a Pi 4 running Bookworm:

```
+ sudo pip install cryptography==38.0.2
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```
You can also see the logs for more details.

[anthias-bookworm-install-error-logs.log](https://github.com/Screenly/Anthias/files/13061564/anthias-bookworm-install-error-logs.log)

From what I've got recently, the script works on Bulllseye (which is now considered legacy), but not on the latest stable release (Bookworm).

The install script needs adjustment.

miichalk, 2023-10-23:
I have the same problem, after installation file screenly is empty.

Napsty, 2023-11-22:
Same here. Brand new Raspberry Pi 4, new install. Log:

```
Anthias requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Do you want Anthias to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/anthias.git
+ '[' '!' -d /home/pi/screenly ']'
+ mkdir /home/pi/screenly
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt update -y
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease        
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
116 packages can be upgraded. Run 'apt list --upgradable' to see them.
+ sudo apt-get install -y --no-install-recommends git libffi-dev libssl-dev python3 python3-dev python3-pip whois
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
libffi-dev is already the newest version (3.4.4-1).
libssl-dev is already the newest version (3.0.11-1~deb12u2+rpt1).
python3 is already the newest version (3.11.2-1+b1).
python3-dev is already the newest version (3.11.2-1+b1).
python3-pip is already the newest version (23.0.1+dfsg-1+rpt1).
whois is already the newest version (5.5.17).
0 upgraded, 0 newly installed, 0 to remove and 116 not upgraded.
+ '[' n == y ']'
+ export MANAGE_NETWORK=false
+ MANAGE_NETWORK=false
+ '[' master = master ']'
++ curl -s https://raw.githubusercontent.com/screenly/anthias/master/requirements/requirements.host.txt
++ grep ansible
+ ANSIBLE_VERSION=ansible-core==2.12
+ sudo pip install cryptography==38.0.2
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
+ sudo pip install ansible-core==2.12
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
[... 61 lines cut ...]
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142) ...
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi4-rpi-v8
update-initramfs: Generating /boot/initrd.img-6.1.0-rpi4-rpi-2712
+ sudo find /usr/share/doc -depth -type f '!' -name copyright -delete
+ sudo find /usr/share/doc -empty -delete
+ sudo rm -rf /usr/share/man /usr/share/groff '/usr/share/info/*' /usr/share/lintian /usr/share/linda /var/cache/man
+ sudo find /usr/share/locale -type f '!' -name en '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' -delete
+ sudo find /usr/share/locale -mindepth 1 -maxdepth 1 '!' -name 'en*' '!' -name 'de*' '!' -name 'es*' '!' -name 'ja*' '!' -name 'fr*' '!' -name 'zh*' '!' -name locale.alias -exec rm -r '{}' ';'
+ sudo chown -R pi:pi /home/pi
+ '[' '!' -f /etc/sudoers.d/010_pi-nopasswd ']'
++ git rev-parse --abbrev-ref HEAD
fatal: not a git repository (or any of the parent directories): .git
++ git rev-parse --short HEAD
fatal: not a git repository (or any of the parent directories): .git
++ lsb_release -a
+ echo -e 'Anthias version: @\nDistributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm'
+ '[' false = false ']'
+ set +x
Installation completed.
You need to reboot the system for the installation to complete. Would you like to reboot now? (y/N)
```

Napsty, 2024-01-05:
Thanks @Doxylamin , your workaround works! But, as you mentioned, this should be fixed in the installation routine, I fully agree.