Screenly/Anthias #1837: No prompt for Wifi when disconnected from ethernet

hanszorn, 2023-10-25:
#### Overview of the Issue

According to your description, if no ethernet cable is connected, Anthias player should offer a Wifi access point to configure wifi, and a screen that tells you about this. However, this does not happen. Instead a screen comes up with 'Internal Server Error (500)' comes up. How can I get around this? Note: with ethernet cable, everything works fine.

#### Reproduction Steps

1. Create a bootable sd card using the Raspbery Pi Imager
2. Boot the pi with this card, with no ethernet cable connected

#### Environment

* Raspberry Pi Hardware Version: 4B
* Raspberry Pi Network Setup: built-in WiFi
* Anthias Version: latest

nicomiguelino, 2023-10-30:
Hello, @hanszorn. Thanks for creating this issue. I've tried creating a prompting mechanism (if disconnected), but it still doesn't work. Still fixing some stability issues. Will come back to this in the future.

As of the moment, you can check [this documentation for manual Wi-Fi setup](https://github.com/Screenly/Anthias/blob/master/docs/wifi-setup.md).

hanszorn, 2023-10-31:
Thanks. Let's hope the future will be soon...