Screenly/Anthias #1838: 💡 Feature Request: Add QR Code Image to Screen for Easy URL Scanning

salmanfarisvp, 2023-11-01:
**Requested Feature**
I would like to propose the addition of a QR code image to the screen of Anthia's application when the URL is displayed as text. This feature would make it convenient for users to scan the QR code and quickly access the URL, allowing for a more seamless and user-friendly experience.

**Additional Considerations**
We should provide an option in the application's settings to enable or disable this feature for users who may prefer not to display the QR code.

andiohn, 2024-07-12:
This is a great idea

andiohn, 2024-07-12:
Another idea is to simply have the URL or localhostname encoded in the QR code, so it could simply be generated.