Screenly/Anthias #1842: youtub-dl to yt-dlp

Sergio1603, 2023-11-29:
Good evening, I tried using Screenly/Anthias with YouTube. I replaced 'youtube-dl' with 'yt-dlp', followed various guides by modifying the 'requirements.txt' and 'utils.py' files. I tested it on different versions of the Raspberry Pi operating system. My only doubt concerns the incorrect reconstruction of Docker after the various modifications. Thank you if you can provide assistance.
(Raspberry pi 4b)

[image]

extremeskillz, 2023-12-02:
I believe they have an open ticket on this. I have the same issue as well. Currently I just download the YT video and upload it to Anthias to play it.

Sergio1603, 2023-12-02:
Hi, I'm doing this too for now

nicomiguelino, 2023-12-11:
Duplicate of 1779

closed 2023-12-11 by nicomiguelino