Screenly/Anthias #1844: Anthias Image from RPi Imager is Unusable

baqwas, 2023-12-07:
#### Overview of the Issue

- What are the parameter values to perform the first login remotely?
  Cannot change _hostname_ and _SSH opt-in_ when changing settings in RPi Imager
- Pointing browser address to RPi 4 (after using Anthias image) does not display anything

#### Reproduction Steps

Steps to reproduce this issue:

* Run RPi Imager (v1.8.1)
* Select Raspberry Pi Device: **RASPBERRY PI 4**
* Select Operating System: **Anthias (pi4)**
* Select Storage: SDDR-113 microSD
* Unable to use Settings options to change hostname and preferred SSH option
* Transferred microSD card to RPi 4 slot
* Booted RPi 4 for the first time
* Launched browser from another computer with known IP address of Anthias server(?)

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4
* Raspberry Pi Network Setup: Ethernet (PoE)
* Anthias Version: **2022-12-12 image** (_via_ RPi Imager)

The RPi 4 works with Bookworm and Bullseye outside this exercise (**before** and **after** Anthias installation attempts)

salmanfarisvp, 2023-12-07:
@baqwas Thank you for reporting the issue, We will investigate further and get back to you.

salmanfarisvp, 2023-12-11:
@baqwas The OS from the Raspberry Imager is too old and it's the same Balena image, thus we can't enable SSH and configure the network. 

I recommend installing the Anthias using the Script method. You can refer to the step-by-setup guide I wrote https://www.hackster.io/Salmanfarisvp/setting-up-anthias-on-raspberry-pi-for-digital-signage-9f6a1b 

please check and let me know if you have any other questions.

Napsty, 2023-12-28:
@salmanfarisvp didn't you run into this issue when trying to run the script method on Raspberry Pi 4? https://github.com/Screenly/Anthias/issues/1834

tdewey-rpi, 2024-01-31:
> @baqwas The OS from the Raspberry Imager is too old and it's the same Balena image, thus we can't enable SSH and configure the network.
>
> I recommend installing the Anthias using the Script method. [...]

Can you please contact me via applications@raspberrypi.com to arrange for an update to the OS List to either remove Anthias or provide an updated image.

Thanks!

Napsty, 2024-01-31:
@baqwas to change SSH settings in this (old) image based on BalenaOS, you can manually adjust the Balena settings. See https://www.claudiokuenzler.com/blog/1378/how-to-enable-ssh-key-authentication-anthias-balenaos-raspberry-pi

salmanfarisvp, 2024-02-22:
> @salmanfarisvp didn't you run into this issue when trying to run the script method on Raspberry Pi 4? #1834

No. It's worked fine in my RPi4 + Ethernet setup.

salmanfarisvp, 2024-02-22:
> @baqwas The OS from the Raspberry Imager is too old and it's the same Balena image, thus we can't enable SSH and configure the network.

There is a correction. The Balena image will automatically update to the latest version once the device is connected to the OS. So, you can flash the OS and connect to ethernet and firmware will update on the go. 

@tdewey-rpi I confirm the above statement and looks working fine. Could you please re-initiate the OS image.

tdewey-rpi, 2024-02-22:
So that I understand correctly:

The Anthias image that was offered in Raspberry Pi Imager mandated connection to the internet in order to be used? Or, per the comment from @Napsty, would require a user to manually edit a schemaless file?

salmanfarisvp, 2024-02-22:
It should work out of the box with the old version as soon it's flashed and connected, and at the same time it will update to the latest os on the go.

Napsty, 2024-02-22:
The comment I added was how to enable SSH access to the Raspberry Pi device, once Anthias image was installed. 
The image itself by default does not allow SSH access (so sad...) - the credentials are unknown to the end user.

vpetersson, 2024-02-22:
If you want that level of control, just do a native installation of Anthias on Raspbian. Having a default SSH open *and* a pre-set credential is a really way to get devices compromised.

nicomiguelino, 2024-03-16:
If you'd like to install Anthias (that runs on Balena), we've created a new release: https://github.com/Screenly/Anthias/releases/tag/v0.18.7.
- From there, you can download the image/s that you need.
- You can flash a micro-SD card with the image using **_balenaEtcher_**.
- You don't need to extract the `.zip` file. You can load it directly on Etcher.
- Please let us know if you still have issues.

baqwas, 2024-03-16:
Thanks, @nicomiguelino. I'll try this image shortly.

Regards.

closed 2024-03-16 by baqwas