Screenly/Anthias #1850: Videos stopped playing on Raspberry Pi4 with BalenaOS install image

Napsty, 2023-12-28:
#### Overview of the Issue

Videos (assets) stopped playing. Only a static Anthias screen is shown on the screen.
Issue seems to be caused by the viewer container and wrong parameters for VLC.

```
root@ba53759:~# balena-engine logs 94abdf380859
vlc: unknown option or missing mandatory argument `--mmal-display=HDMI-2'
Try `vlc --help' for more information.
Traceback (most recent call last):
  File "viewer.py", line 73, in <module>
    media_player = VLCMediaPlayer() if 'Raspberry Pi 4' in get_raspberry_model() else OMXMediaPlayer()
  File "/usr/src/app/lib/media_player.py", line 37, in __init__
    self.player = self.instance.media_player_new()
AttributeError: 'NoneType' object has no attribute 'media_player_new'
Home directory not accessible: Permission denied
[02043388] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Home directory not accessible: Permission denied
[020486a8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Loading browser...
Scheduler init
update_playlist
Generating asset-list...
generate_asset_list deadline: 2023-12-27 14:32:00
update_playlist done, count 1, counter 0, index 0, deadline 2023-12-27 14:32:00
Starting new HTTP connection (1): anthias-nginx:80
http://anthias-nginx:80 "GET / HTTP/1.1" 200 None
Starting new HTTP connection (1): 10.114.104.1:48484
http://10.114.104.1:48484 "GET /v1/device?apikey=04bfe54bf1234285aeec7e19729b5f50 HTTP/1.1" 200 322
Current url is http://anthias-nginx:80/splash-page
```

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Add Assets
* Press "On"
* Worked for a couple of days, then after uploading and enabling new assets, stopped working (for all assets)

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.5
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter): Ethernet, static DHCP reservation
* Anthias Version: master@b109dc1 using BalenaOS install image

Napsty, 2023-12-28:
Errors in viewer container logs, when I manually click on "Next Asset":

```
[h264 @ 0xe42188c0] no frame!
get_next_asset
refresh_playlist
refresh: counter: (0) deadline (2023-12-27 14:32:00) timecur (2023-12-28 15:36:31.063231)
update_playlist
Generating asset-list...
generate_asset_list deadline: 2023-12-27 14:32:00
get_next_asset after refresh
get_next_asset counter 0 returning asset 1 of 1
Showing asset NeueLieblingsbrillefuer2024.mp4 (video)
Asset URI /data/screenly_assets/5fce0283701f4705882b73171022febf.mp4
Displaying video /data/screenly_assets/5fce0283701f4705882b73171022febf.mp4 for 15 
Reading config-file...
Current url is null
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
[f1ec05d0] main video output error: video output creation failed
[e440bd40] main decoder error: failed to create video output
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
[f1ec05d0] main video output error: video output creation failed
[e440bd40] main decoder error: failed to create video output
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
[e46d0998] main video output error: video output creation failed
[e440bd40] main decoder error: failed to create video output
[h264 @ 0xe4401900] get_buffer() failed
[h264 @ 0xe4401900] thread_get_buffer() failed
[h264 @ 0xe4401900] decode_slice_header error
[h264 @ 0xe4401900] no frame!
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
```

salmanfarisvp, 2024-01-01:
Hi @Napsty, The Raspberry Pi imager has an old version of the Anthias Balena image. For the newest version, For the newest version please use the Anthias installation using the Script or follow the guide here [Setting Up Anthias on Raspberry Pi for Digital Signage 🖥️ ](https://www.hackster.io/Salmanfarisvp/setting-up-anthias-on-raspberry-pi-for-digital-signage-9f6a1b), Please check and let us know. 

The issue you mentioned was fixed in the latest PR https://github.com/Screenly/Anthias/pull/1849

Napsty, 2024-01-04:
Will try and report back

Napsty, 2024-01-05:
Hi @salmanfarisvp I tried to install Anthias using the script method and following your tutorial, but it doesn't work. The setup script runs into an error when trying to install the Ansible package.

Excerpt:

```
...
Processing triggers for man-db (2.11.2-2) ...
+ '[' Y == y ']'
+ export MANAGE_NETWORK=false
+ MANAGE_NETWORK=false
+ '[' master = master ']'
++ curl -s https://raw.githubusercontent.com/screenly/anthias/master/requirements/requirements.host.txt
++ grep ansible
+ ANSIBLE_VERSION=ansible-core==2.12
+ sudo pip install cryptography==38.0.2
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
+ sudo pip install ansible-core==2.12
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
+ sudo -u pi ansible localhost -m git -a 'repo=https://github.com/screenly/anthias.git dest=/home/pi/screenly version=master force=no'
sudo: ansible: command not found
+ cd /home/pi/screenly/ansible
/dev/fd/63: line 190: cd: /home/pi/screenly/ansible: No such file or directory
+ sudo -E -u pi ansible-playbook site.yml
sudo: ansible-playbook: command not found
+ sudo -u pi /home/pi/screenly/bin/upgrade_containers.sh
sudo: /home/pi/screenly/bin/upgrade_containers.sh: command not found
+ sudo apt-get autoclean
...
```

That's the issue https://github.com/Screenly/Anthias/issues/1834 hitting here.

salmanfarisvp, 2024-01-05:
@Napsty Can you confirm the OS version and is it a 32 or 64-bit OS? and how did you connect to the internet?

Napsty, 2024-01-05:
One needs to apply the workaround mentioned in https://github.com/Screenly/Anthias/issues/1834#issuecomment-1850181200 to be able to install Anthias using the script method (at least with Raspbian Bookworm 64bit).

I will test with this new Anthias setup and see whether the videos stop working or not.

salmanfarisvp, 2024-01-05:
@Napsty Bookworm support is still ongoing and you can find the details here at https://github.com/Screenly/Anthias/pull/1836 . Meanwhile, Can you please try to use **Bullseye 64-bit** and let us know?

Napsty, 2024-01-05:
> Bookworm support is still ongoing

Ah that's why, thanks for the information. Status quo: I have a working Anthias environment running on Raspberry Pi 3b with Raspbian Bookworm 64-bit. The containers are all up. I will try with this and compare to the one running with BalenaOS. Keep you updated here.

nicomiguelino, 2024-04-02:
@Napsty, changes to make installer work on Bookworm were recently merged &mdash; https://github.com/Screenly/Anthias/pull/1836