Screenly/Anthias #1851: Black screen between videos on Raspberry Pi 3B using it on Raspberry Pi OS 64 Bit Lite

1amrishi, 2023-12-30:
Installed Anthias on my Raspberry pi 3B after installing the latest Rpi OS 64 Bit Lite as i also wanted to use tailscale for remote access but the issue is that after uploading multiple videos compressd using handbrake, as those are the only videos seeming to work on it, there is a black screen of 4-5 Seconds between each video.

 is this a know or unsolvable issue, please guide me. 

* Raspberry Pi Hardware Version: Raspberry Pi 3B 1GB Ram
* Raspberry Pi Network Setup:built-in WiFi
* Anthias Version: 0.18.6

Napsty, 2024-01-04:
Same issue here, even with just one asset active there's an annoying black screen of around 5 second until the video is played again.

vpetersson, 2024-01-04:
Important variables:

* What kind of SD card (anything less than a Class 10 should be avoided)
* How long is the video?

Napsty, 2024-01-04:
In my case SD card is a class 10, the video duration is 29s.

vpetersson, 2024-01-04:
Does the duration decrease if you use a shorter asset?

Napsty, 2024-01-04:
I was unable to test this myself but the person on site confirmed that the black screen duration is always ~5s, doesn't matter which video file was playing and which duration that video had.

1amrishi, 2024-01-05:
> Important variables:
> 
> * What kind of SD card (anything less than a Class 10 should be avoided)
> * How long is the video?

I am also using a sandisk 32gb class 10
i have multiple videos of 30sec- 1 min

1amrishi, 2024-01-05:
> Does the duration decrease if you use a shorter asset?

no

Napsty, 2024-01-05:
Just installed a new Anthias using the script version on a separate device (Raspberry Pi 3b+) and there's 7-8 seconds black screen between playing the assets.

Napsty, 2024-01-07:
Re-installed the whole thing, using a Raspbian Legacy (Bullseye) 64 bit this time (was Bookworm before). Same thing happens; roughly 8s black screen during video play.

pacmac, 2024-01-29:
Same thing with fresh install. Class 10 card, 29 second long video, blank screen for about 3 seconds at the end / beginning

nicomiguelino, 2024-06-13:
@ShubhamGhait
- As of now, setup/customization of screen resolution is not supported.
- How long is the black screen in your device when playing videos?

gram11111111, 2024-06-23:
I still get a black screen between videos it's around 4 seconds, installed on a Pi 3B+ if it has been running for a bit the transitions get even longer

nicomiguelino, 2024-06-24:
@gram11111111, thanks for sharing your findings.
* Video assets were played using OMX Player.
* I'm still assessing and trying out alternatives to OMX Player, like VLC and Qt (via QMediaPlayer).

gram11111111, 2024-06-24:
Thank you for all your contribution, really hope you'll be able to fix it

ShubhamGhait, 2024-07-05:
> @ShubhamGhait
> 
> * As of now, setup/customization of screen resolution is not supported.
> * How long is the black screen in your device when playing videos?

Sorry for the wait for the response.
1. There is a black screen for about four to five seconds every video.
Should you possess the ability to resolve the aforementioned issue, please offer a remedy.

nicomiguelino, 2024-07-05:
@ShubhamGhait, I'm still on the phase of making the playback work with using Qt (instead of OMX) for playing videos.
Based on testing the changes (still a work in progress), the black screen between videos seems to disappear, but there's the risk of cross-play (where one video plays where the other video is still playing).

If there's no progress in this one, I'll be proceeding with trying out things VLC again.

Nezzu-bit, 2024-07-23:
@nicomiguelino 

Whats the progress till now? i got the same issue. Blackscreen of 6-8 seconds between every video on my Pi 4b. Doesnt matter which video format im using ( .mov, .mp4, .avi, .mkv... ). Are there any fixes of that?

nicomiguelino, 2024-07-23:
@Nezzu-bit, changes (migration from OMX to VLC) were aleady merged and the deployment is underway.

Chrisong81, 2024-09-25:
I use a COM4 (Compute Module 4) with 2GB RAM and a 32GB Class 10 memory card. I installed Screenly Anthias, and it plays images, but when switching to video, the screen goes blank before showing the IP address page. Does anyone know the problem?

nicomiguelino, 2024-10-04:
@Chrisong81, I'd like to ask for the following details:
* Anthias Version
* OS (Raspberry Pi OS Lite or BalenaOS)
* Viewer logs, which can be retrieved by running the following:

  ```bash
  cd ~/screenly
  docker compose logs -ft anthias-viewer
  ```

Thanks!

alissonjcguedes, 2025-03-04:
> [@Chrisong81](https://github.com/Chrisong81), I'd like to ask for the following details:
> 
> * Anthias Version
> * OS (Raspberry Pi OS Lite or BalenaOS)
> * Viewer logs, which can be retrieved by running the following:
>   cd ~/screenly
>   docker compose logs -ft anthias-viewer
> 
> Thanks!

Hello!

Device
Raspberry Pi 3 Model B Rev 1.2

Anthias
[master@02da7f6]

OS
[balenaOS 6.1.24+rev2]

Log
 anthias-viewer  Generating asset-list...
 anthias-viewer  Showing asset Huga Anuncie Aqui (1920 x 1080 px) - Corrigido.mp4 (video)
 anthias-viewer  Current url is null
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  Fontconfig error: No writable cache directories
 anthias-viewer  error: XDG_RUNTIME_DIR is invalid or not set in the environment.
 anthias-viewer  error: XDG_RUNTIME_DIR is invalid or not set in the environment.
 anthias-viewer  [73204ad0] gles2 generic error: parent window not available
 anthias-viewer  [73204ad0] xcb generic error: window not available
 anthias-viewer  [7325d768] gles2 vout display error: parent window not available
 anthias-viewer  [7325d768] gl vout display error: parent window not available
 anthias-viewer  [7325d768] xcb vout display error: window not available
 anthias-viewer  [7325d768] drm_vout vout display: <<< OpenDrmVout: DPV0 1920x1080(1920x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1920x1080(1920x1080 @ 0,0 1/1)
 anthias-viewer  [7325d768] drm_vout vout display error: Failed to get xlease

nathansensi, 2025-04-23:
Same issues, error log shows the same

hias-viewer-1  | 2025-04-23T17:47:38.392552216Z error: XDG_RUNTIME_DIR is invalid or not set in the environment.
anthias-viewer-1  | 2025-04-23T17:47:38.392890117Z error: XDG_RUNTIME_DIR is invalid or not set in the environment.
anthias-viewer-1  | 2025-04-23T17:47:38.402927178Z [014801d0] gles2 generic error: parent window not available
anthias-viewer-1  | 2025-04-23T17:47:38.403837903Z [0148e2a8] xcb generic error: window not available
anthias-viewer-1  | 2025-04-23T17:47:38.403874661Z [014ac9a0] gles2 vout display error: parent window not available
anthias-viewer-1  | 2025-04-23T17:47:38.404395911Z [014ac9a0] gl vout display error: parent window not available
anthias-viewer-1  | 2025-04-23T17:47:38.405349153Z [014ac9a0] xcb vout display error: window not available
anthias-viewer-1  | 2025-04-23T17:47:38.406290618Z [014ac9a0] drm_vout vout display: <<< OpenDrmVout: DPV0 1920x1080(1920x1080 @ 0,0 0/0), cfg.display: 1920x1080, source: 1920x1080(1920x1080 @ 0,0 1/1)
anthias-viewer-1  | 2025-04-23T17:47:38.406333450Z [014ac9a0] drm_vout vout display error: Failed to get xlease
anthias-viewer-1  | 2025-04-23T17:47:38.406347098Z [014ac9a0] drm_vout vout display error: Failed to set atomic cap
anthias-viewer-1  | 2025-04-23T17:47:38.421670546Z [014ac9a0] drm_vout vout display error: Failed to find output <auto>: Invalid argument

Onltor, 2025-07-29:
Dear developers and users,

I would like to draw your attention to one important detail that no one has mentioned: **before each video starts, the player (or the procedure calling it) inexplicably reinitializes the HDMI mode**, even though it always sets the same mode.

This leads to the following issues:

1. The output device also reinitializes its display mode, showing notifications about the change (only professional media screens more or less smooth out these initializations and suppress the mode-change messages).
2. The “black gap” may be caused precisely by that interruption for HDMI initialization.
Before the v0.19.6 release, the black gap also appeared between still images—was HDMI being reinitialized before each image back then?

Not a single well-known commercial or freemium content player resets the HDMI mode before every video; they do it only once at startup!

Please review the code — I hope this remark will help the community identify and fix the problem more quickly.

Thanks to everyone involved in the project!

Username56454, 2026-01-16:
Hi,

We are experiencing the same issue. The type of display seems to be a factor. During our tests on a standard computer monitor, we had no issues. However, when we connected our Anthias device to the NEC display, we immediately saw a 4-second black screen between each asset, no matter the asses size or type.

We also noticed that Anthias seems to restart the video connection between assets on the NEC display, which we believe is part of the problem.

We have tested the following configurations:

Anthias on Raspberry Pi 3
anthias on Lenovo Tiny PC
Swapping HDMI cables
Using DisplayPort instead of HDMI

The issue occurred in all scenarios involving the NEC displays, but in no scenarios involving a standard computer monitor.

We used the same NEC display with Chrome/Google workspace digital signage and other devices with no issues in the past.