Screenly/Anthias #1853: Any plans to upgrade to Raspberry Pi5 ?

tyronemc2, 2024-01-15:
Hi Guys, 
Just checking if there are any plans to upgrade for Raspberry Pi5 ?

#### Overview of the Issue

Installed from github on Raspberry Pi5 , and the script does not complete the installation.

#### Environment

* Raspberry Pi Hardware Version: 5
* Raspberry Pi Network Setup:Ethernet
* Anthias Version:

vpetersson, 2024-01-15:
There are plans for it. We just haven't been able to get to it yet due to some other proprieties.

palmarcell, 2024-01-24:
I bought a Raspberry Pi 5 just for this project and now realized, that this is a known issue... 😅 
@vpetersson is there any roadmap for this update? Is there any workaround?

vpetersson, 2024-01-24:
@salmanfarisvp is looking at it. Unfortunately the pi5 changed things a lot so under the hood so it might require a fair bit of conditional support needed to be added.

palmarcell, 2024-01-24:
Thanks a lot! Then I think I need to downgrade to RPI 4 till that...

bozzer91, 2024-03-25:
Good evening all just wondering if there is a rough time frame yet for Pi 5 compatibility?

vpetersson, 2024-03-26:
No stated timeline, but contributions are welcome.

dknf, 2024-03-28:
Hello, we bought raspberry pi 5 to do digital signage in our university, we are looking forward to your solution :)

rapazdaluta, 2024-06-29:
> Hello, we bought raspberry pi 5 to do digital signage in our university, we are looking forward to your solution :)

I've also bought pi 5 and only now I see this is a known issue hehe
Hopefully Anthias will be supported by Pi5 devices this year's end 👍

drewhemm, 2024-06-29:
I was impatient and bought a Pi 4 B rather than wait for the 5 to be supported. It's the same case with most digital signage software unfortunately.

vpetersson, 2024-06-29:
The reason being that the way video is processed changed in the Pi 5 (for the better IMHO). That's why few signage providers support it today. 

We're hoping to add support in Screenly (and Anthias), but depending on the graphics stack it can be a fair bit of work.

marcinmikaGL, 2024-12-14:
Hello,
As we are approaching the end of the year, are there any chances of Anthias being available for Raspberry Pi 5 in the near future?

nicomiguelino, 2024-12-20:
@marcinmikaGL
Anthias now supports Raspberry Pi 5 devices running Raspberry Pi OS Lite (64-bit Bookworm). More details can be found in #1868.

We'll be adding support for Pi 5 devices running balenaOS in the future.

closed 2024-12-20 by nicomiguelino