Screenly/Anthias #1866: jQuery version 1.9.1 security vulnerabilities

M4rkWilhelm, 2024-03-25:
#### Overview of the Issue
The version of jQuery imported in the head.html template is the subject of multiple CVEs.

CVE-2015-9251
CVE-2020-11023
CVE-2019-11358
CVE-2020-11022

Migrating to jQuery 3.7 would be prudent.

#### Reproduction Steps

View source or open developer console and inspect header of any Anthias backend page.

Note the version of the minified jQuery script import.

vpetersson, 2024-03-26:
@nicomiguelino please pick this up next gap week.

M4rkWilhelm, 2024-03-26:
Some tools to assist in the migration:

- (Guide) https://jquery.com/upgrade-guide/3.0/
- (1.x/2.x migration plugin) https://github.com/jquery/jquery-migrate/tree/1.x-stable/
- (3.x/4.x migration plugin) https://github.com/jquery/jquery-migrate/

nicomiguelino, 2024-03-26:
@M4rkWilhelm Cool, thanks! I'll pick this up next week.

PR #1875 (2024-04-05): Fixes jQuery security vulnerabilities
#### Overview

* There are no relevant changes yet.
* Fixes #1866

#### Prep work

I can move this to a separate PR as needed.

* Install Node.js when the server's started in development mode.
* Installs Node.js dependencies (like CoffeeScript) for a smoother workflow.
closed 2024-04-05 by nicomiguelino