Screenly/Anthias #1874: Feature request - A must. Option to schedule display period on daily basis

lovrenco, 2024-04-05:
I cannot believe this is not available till now. 

In short, i need to show different guides on the screen, for different parts of the day.

For example:
8:00 till 14:00 - Photo1 
14:00 till 18:00 - Photo2
18:00 till 24:00 - Photo3
00:00 till 8:00 - Some default screen, - Photo4

In current setup this is not possible? Why is that? It is simple CRON function.
Is there a chance for this to be added soon?

nicomiguelino, 2024-04-05:
@lovrenco, thanks! In the meantime, if you'd like to do some digging, you can check [`viewer.py`](https://github.com/Screenly/Anthias/blob/master/viewer.py).
- You can start browsing on the `asset_loop(scheduler)` function, then step into `get_next_asset()`.

lovrenco, 2024-04-06:
@nicomiguelino Thanks. I checked the code, and I can see the logic, but unfortunately I am not developer. 

Do you think function I suggested could be added soon? If not I will have to search for another solution.

vpetersson, 2024-04-08:
It's already available in [Screenly](https://www.screenly.io), but it's not currently on the roadmap for Anthias.

lovrenco, 2024-04-10:
@vpetersson Sorry to hear that. It is such a basic functionality, but I can see that it might compete with your paid product that way.