Screenly/Anthias #1878: Blinking top left corner

happen-studio, 2024-04-08:
#### Overview of the Issue

On the screen, in the top left corner I get a blinking filed, looks like a terminal cursor to me. 
Anyone here, who had the same bug? Any idea to get rid of it?

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.5
* Raspberry Pi Network Setup: built-in WiFi,
* Anthias Version: master@7b559ef

THX

[image]

nicomiguelino, 2024-04-08:
If it only occurs on devices running Bookworm, then it could be that the splash screen settings are not being set properly.

Do you happen to have one installed via the Imager or via the release images (both of which are Balena-based images)?

joshbenner, 2024-04-15:
I see the same behavior installed on Bookworm. I've tested with Balena-based images and did not notice this behavior.

nicomiguelino, 2024-04-15:
Thanks for sharing the results on your end. Does anyone experience similar issue on Bullseye.

4xel, 2024-04-23:
[image]
I have the same issue, fresh install of Raspberry PI OS using 'imager' and install via instructions in readme. No network management, yes to updates

Monitor Info	state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
Raspberry Pi Model	Raspberry Pi 4 Model B Rev 1.5
Anthias Version	master@7b559ef

PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm

Can I resolve this by downgrading?

nicomiguelino, 2024-04-23:
You can use Bullseye instead, if it does work.

closed 2024-05-14 by nicomiguelino