Screenly/Anthias #1879: Certain Websites Crash into White Screen. Websites do not continuously play.

ryanmilla, 2024-04-10:
Issue: Certain websites crash into a blank white screen and do not want to continue to be displayed. 

Websites are up for a specific amount of time and crash into a blank white screen for a specific amount of time as well. 

1. I have this website (internal to my network. it's a slideshow) and it plays for 30 seconds and crashes into this blank white screen for 8 minutes. It is the only asset that is enabled.
2. https://www.bbc.co.uk/news shows for half a second and crashes into the same white screen for 5 minutes. It is the only asset that is enabled.

These websites also do not offer smooth transitions.
3. When one of these websites are enabled with the addition to a simple .jpg with a logo on it, it tends to hang on this white screen for a second before the website shows its contents. It then proceeds to crash into this white screen as mentioned in 1. and 2.

#### Environment
* Raspberry Pi Hardware Version: RPI 4B, 2GB RAM, Class 10 microSD Card with Raspberry Pi Lite OS 64GB, USB-C 5.1V, 3.0A Power Supply
* Raspberry Pi Network Setup: Ethernet
* Anthias Version: Latest Release

nicomiguelino, 2024-04-10:
@ryanmilla Thanks for creating this issue.
Do you use a larger display (> 1920x1080)?
- I don't experience the issue on my FHD monitor, but experienced the same months ago on a 4K display.
- Let me try to test it again on a larger display.

nicomiguelino, 2024-04-10:
@ryanmilla, but things get pretty unusual when I unplugged from my FHD monitor and plugged to my 4K, without device reboot. The larger display can still show the BBC page. Should you wanna do some digging too, you can also do research on how to set resolution on the fly using `tvservice`.

`tvservice` can be accessed by going inside the `anthias-server` container:

```bash
cd ~/screenly
docker compose exec anthias-server bash

# Then run the `tvservice` command

tvservice -m CEA # lists all the possible groups for CEA mode
tvservice -m DMT # lists all the possible groups for DMT mode

tvservice -e `CEA <GROUP_ID>`
```

Still can't make it work to display in 1920x1080 for larger (4k) displays.

ryanmilla, 2024-04-10:
@nicomiguelino Thanks for the response. I appreciate it.

This issue occurs all FHD displays and anything bigger than that as well. 

You've mentioned that I wouldn't be able to run these commands. What alternative is there for Anthias instances isntalled on RPi OS Lite?

nicomiguelino, 2024-04-10:
@ryanmilla, I thought you're using a pre-release image 'cause it's mentioned that you're using 0.18.7, but I overlooked the part where the device runs RPi OS Lite.If that's the case, you should be fine and should be able to run the commands.

Don't have a possible solution yet, but can you provide a screenshot of your **_System Info_** here? (Just make sure not to disclose any info you think are sensitive.)

ryanmilla, 2024-04-10:
[image]

Hi. It's currently plugged into a 4K monitor. I've also tested it on a regular 1920x1080 FHD monitor and its the same result.

Thanks.

nicomiguelino, 2024-04-10:
@ryanmilla, thanks.
- I'll use that info in debugging this issue.
- I'm testing it on a Pi 4 too, but mine has 4GB RAM.
- During installation, the script modifies the GPU memory (`gpu_mem`) allocated.
- For your device (that has 2GB RAM), `gpu_mem` will be set to `196`.
- You can try tweaking `boot/firmware/config.txt` and set `gpu_mem_1024` to something higher to see if that works.

ryanmilla, 2024-04-10:
Hi @nicomiguelino 

Changing CEA mode didn't work. 

Hi, my config file was in /boot/config.txt.
I set gpu_mem_1024=256 and that didn't seem to work either.

Thanks for the suggestions so far.

ryanmilla, 2024-04-11:
Hi @nicomiguelino Just ran into the same issue you've run into. When plugging into a different 1920x1080, https://www.bbc.co.uk/news displays. Once I unplug it (without rebooting) and put it onto a 4k display, it shows up as well. I wonder if there's a way for "monitor info" to be kept static during boot up?

Working (booted from 1920x1080 then plugged into 4K)
[image]

Not working (booted straight from 4k display)
[image]

ryanmilla, 2024-04-11:
@nicomiguelino Hi Nico. I have the fix.

Given your CEA mode list, certain websites will crash if it isn't tied to the right CEA mode. https://www.bbc.co.uk/news enjoys CEA mode 31. 

Persistent CEA modes can be set in the config.txt file. /boot/config.txt
hdmi_group=1 #turns on persistent CEA groups
hdmi_mode=31 #https://www.bbc.co.uk/news likes this CEA mode.

You may close the issue. Thank you.

nicomiguelino, 2024-04-11:
Thanks, @ryanmilla! I'm going to try the steps you did on my Pi before closing it.
In the meantime, I'll include this issue (as a task) in this repo's project.
I'll might add docs for dealing with such cases.

ryanmilla, 2024-04-11:
@nicomiguelino Thanks a ton for the help!

nicomiguelino, 2024-04-12:
@ryanmilla, so I recently tested the config changes. The device still displays a blank, white screen on my 4K TV for the BBC web page. I checked the **_Monitor Info_** and it still displays the following:

```
state 0xa [HDMI CUSTOM RGB lim unknown AR], 4096x2160 @ 30.00Hz, progressive
```

ryanmilla, 2024-04-12:
@nicomiguelino my config works for 4K monitors too. I’m currently using CEA mode 4 but that’s because I’m dealing with a pretty finicky website. 

The issue that I’m now having is forcing the pi to always boot into CEA mode 4 (or in the case of BBC web page: CEA mode 31). Sometimes the config pushes it, sometimes it doesn’t.

FYI I’m editing /boot/config.txt if that makes a difference. I’ll show you a screenshot of the config in a few hours. Thanks

ryanmilla, 2024-04-12:
@nicomiguelino I have multiple pi's and some of them do not want to push my static CEA mode when its plugged in. Sometimes it just reverts back.

This is my config for /boot/config.txt
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
hdmi_group=1
hdmi_mode=4

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=10

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
#dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
dtoverlay=vc4-fkms-v3d
# Run as fast as firmware / board allows
arm_boost=1

[all]
framebuffer_depth=32
framebuffer_ignore_alpha=1
gpu_mem_256=96
gpu_mem_512=128
gpu_mem_1024=196

ryanmilla, 2024-04-12:
@nicomiguelino try adding this line to the config.txt file:
 `hdmi_ignore_edid=0xa5000080`

I believe it forces to TV to take the Pi’s preferred CEA mode

ryanmilla, 2024-04-12:
@nicomiguelino This entire issue may be a player issue. The player just doesn't communicate well with the TV. In some cases, I can get CEA mode 31 working on some 1920x1080 monitors & no 4K monitors.

Cheers.

ryanmilla, 2024-04-18:
@nicomiguelino just an FYI: tested other sites and 1920x1080 CEA modes still result in a white screen. I've tried changing the player from OMX to VLC and it didn't work out. Cheers.

nicomiguelino, 2024-04-18:
@ryanmilla Thanks. That's unfortunate. It could be that the Qt/C++ needs fixing. There are plans to upgrade from Qt 5 to Qt 6, then made the necessary adjustments to the C++ code.

ryanmilla, 2024-04-21:
@nicomiguelino Thanks for the help. I'll be waiting for that update then! Cheers

nicomiguelino, 2024-05-16:
@ryanmilla, Qt-related changes hasn't been started yet, but I tweaked relevant Docker Compose files in hopes of resolving the white screen issue. Here's the PR for reference: #1894, which includes fix for playback of web assets.

Tweaking the [shm_size](https://docs.docker.com/compose/compose-file/05-services/#shm_size) of the `anthias-viewer` service helped. The changes are already merged, and you can try to test the changes.

For your device running Anthias on Bookworm, you can run the following, but make sure that you've backup your data.

```bash
# Make sure that you're inside the project root directory.
cd ~/screenly

# Pull the latest changes from the `master` branch.
git pull 

# If you can't pull the changes, you can just add a `shm_size` in the `anthias-viewer` service of `docker-compose.yml.tmpl`.
# You can set it to anything larger than `64mb` (which is the default value.) like `256mb`, `1gb`, etc.

# Restart the docker containers, but you have to do it in a specific way, since `docker-compose.yml` is auto-generated based on the template file `docker-compose.yml.tmpl`.
docker compose down && ./bin/upgrade_containers.sh
```

closed 2024-05-20 by nicomiguelino

ryanmilla, 2024-06-16:
@nicomiguelino Hi Nico.

Sorry for the late response. Tested this after a month. Works a lot better now. I'd like to thank you for your time.

Regards,
Ryan.

ryanmilla, 2024-07-09:
Hi @nicomiguelino really enjoying the update. 

However, how do I force a CEA mode? I've seen that you've moved the config.txt file to /boot/firmware/config.txt but entries are CEA mode are no longer in the config.txt file

Thank you

nicomiguelino, 2024-07-09:
@ryanmilla, I haven't dug deep into the new config parameters. `hdmi_group` and `hdmi_mode` are now considered [legacy options](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#legacy-conditional-filters).

I found a [documentaton](https://www.raspberrypi.com/documentation/computers/configuration.html#manually-set-resolution-and-rotation) about manually setting resolution, but it seems that it's only applicable for those using Wayland.

vpetersson, 2024-07-10:
This is the way. Everything is moving towards Wayland. We need to update the Anthias stack to be Wayland first as the old way of doing things (`config.txt` etc) is being sunsetted by the Pi Foundation.

ryanmilla, 2024-07-21:
@nicomiguelino @vpetersson Thanks guys, I appreciate it.