Screenly/Anthias #1887: Migrate WebView from Qt 5 to Qt 6.

nicomiguelino, 2024-04-22:


palmarcell, 2024-04-22:
Hi @nicomiguelino,
Do you have any estimate about the time effort of this? I saw that you created this issue but put On hold right away.
As I see in the Issues tab the `white screen for 4K screens` is a somewhat common problem among the tickets.

nicomiguelino, 2024-04-22:
@palmarcell, I'm sorry about that. I should be putting other issue On Hold, not this one.
Worst case scenario, it might take a week or two, taking into account new dependencies (if there are any) when switching to Qt 6.

palmarcell, 2024-04-22:
Thanks for the quick reply and contribution!
Let's hope this will fix the issue! :)

nicomiguelino, 2024-04-22:
@palmarcell, I backtracked and forgot to specify that I meant 2-3 weeks of full time work.
I currently have other responsibilities, including other Anthias issues.

Contributions are always welcome.

palmarcell, 2024-05-16:
@nicomiguelino , thanks for your changes in https://github.com/Screenly/Anthias/pull/1894 !
I have upgraded my containers and the app works flawlessly with 4k!