Screenly/Anthias #1891: Install doesn't work. Stop at TASK [system : Install Docker]  on Raspbery pi zero 2W

idMasjid, 2024-04-27:
Hello,

I have a fresh raspberry os lite 64bit'
Last command before run script
sudo apt update
sudo apt full-upgrade
sudo rpi-update
sudo reboot
And run script
$ bash <(curl -sL https://install-anthias.srly.io)
Install doesn't work. Stop at TASK [system : Install Docker]  on Raspbery pi zero 2W.
How can i install Anthias on raspberry pi Zero 2W

Raspberry Pi Zero 2W
Raspberry Pi Network Setup: built-in WiFi
Anthias Version the last version
 
 Install log: 
`
       d8888            888     888
      d88888            888     888       888
     d88P888            888     888
    d88P 888  88888b.   888888  88888b.   888   8888b.   .d8888b
   d88P  888  888 '88b  888     888 '88b  888      '88b  88K
  d88P   888  888  888  888     888  888  888  .d888888  'Y8888b.
 d8888888888  888  888  Y88b.   888  888  888  888  888       X88
d88P     888  888  888   Y888   888  888  888  'Y888888   88888P'
==================================================================

Anthias requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Do you want Anthias to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/anthias.git
+ '[' '!' -d /home/minus/screenly ']'
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
++ lsb_release -rs
+ RASPBIAN_VERSION=12
+ APT_INSTALL_ARGS=("git" "libffi-dev" "libssl-dev" "whois")
+ '[' 12 = 12 ']'
+ APT_INSTALL_ARGS+=("python3-full")
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt update -y
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:5 https://download.docker.com/linux/debian bookworm InRelease
Fetched 55.4 kB in 3s (17.8 kB/s)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: https://download.docker.com/linux/debian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
+ sudo apt-get install -y --no-install-recommends git libffi-dev libssl-dev whois python3-full
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
libffi-dev is already the newest version (3.4.4-1).
libssl-dev is already the newest version (3.0.11-1~deb12u2+rpt1).
whois is already the newest version (5.5.17).
python3-full is already the newest version (3.11.2-1+b1).
The following package was automatically installed and is no longer required:
[... 147 lines cut ...]

TASK [system : Install libc6-dev] ****************************************************
changed: [localhost]

TASK [system : Install Screenly dependencies] ****************************************
ok: [localhost]

TASK [system : Remove deprecated apt dependencies] ***********************************
ok: [localhost]

TASK [system : Make sure distro package of Docker is absent] *************************
ok: [localhost]

TASK [system : Add docker apt key] ***************************************************
ok: [localhost]

TASK [system : Get raspbian name] ****************************************************
ok: [localhost]

TASK [system : Add Docker repo] ******************************************************
ok: [localhost]

TASK [system : Install Docker] *******************************************************

`

nicomiguelino, 2024-05-01:
Have you tried installing Docker manually (just do see what specifically the issues are)? These might help:
- [Install Docker Engine on Raspberry Pi OS (32-bit)](https://docs.docker.com/engine/install/raspberry-pi-os/)
- [Install Docker Engine on Debian](https://docs.docker.com/engine/install/debian/)

idMasjid, 2024-05-02:
> Have you tried installing Docker manually (just do see what specifically the issues are)? These might help:
>
>     * [Install Docker Engine on Raspberry Pi OS [...]

after install docker engine on pi os 64 bit the same problem
i  try to renew install the same error with frezze "overload ram"on the step:
       d8888            888     888
      d88888            888     888       888
     d88P888            888     888
    d88P 888  88888b.   888888  88888b.   888   8888b.   .d8888b
   d88P  888  888 '88b  888     888 '88b  888      '88b  88K
  d88P   888  888  888  888     888  888  888  .d888888  'Y8888b.
 d8888888888  888  888  Y88b.   888  888  888  888  888       X88
d88P     888  888  888   Y888   888  888  888  'Y888888   88888P'
==================================================================

Anthias requires a dedicated Raspberry Pi / SD card.
You will not be able to use the regular desktop environment once installed.

Do you still want to continue? (y/N)
Do you want Anthias to manage your network? This is recommended for most users because this adds features to manage your network. (Y/n)

Would you like to perform a full system upgrade as well? (y/N)
+ REPOSITORY=https://github.com/screenly/anthias.git
+ '[' '!' -d /home/root/screenly ']'
+ mkdir /home/root/screenly
+ sudo mkdir -p /etc/ansible
+ echo -e '[local]\nlocalhost ansible_connection=local'
+ sudo tee /etc/ansible/hosts
+ '[' '!' -f /etc/locale.gen ']'
++ lsb_release -rs
+ RASPBIAN_VERSION=12
+ APT_INSTALL_ARGS=("git" "libffi-dev" "libssl-dev" "whois")
+ '[' 12 = 12 ']'
+ APT_INSTALL_ARGS+=("python3-full")
+ sudo sed -i s/apt.screenlyapp.com/archive.raspbian.org/g /etc/apt/sources.list
+ sudo apt update -y
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 https://download.docker.com/linux/debian bookworm InRelease
Hit:5 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
+ sudo apt-get install -y --no-install-recommends git libffi-dev libssl-dev whois python3-full
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
git set to manually installed.
The following additional packages will be installed:
  2to3 blt fonts-mathjax idle idle-python3.11 libjs-mathjax libpython3.11-testsuite libtcl8.6 libtk8.6 libxft2 libxrender1 libxss1 python3-gdbm python3-tk
  python3.11-full tk8.6-blt2.5 x11-common
Suggested packages:
  blt-demo fonts-mathjax-extras fonts-stix libjs-mathjax-doc libssl-doc tcl8.6 tk8.6 python3-dev python3-gdbm-dbg tix python3-tk-dbg python3.11-dev
Recommended packages:
  python3-doc python3-examples python3.11-doc python3.11-examples
The following NEW packages will be installed:
  2to3 blt fonts-mathjax idle idle-python3.11 libffi-dev libjs-mathjax libpython3.11-testsuite libssl-dev libtcl8.6 libtk8.6 libxft2 libxrender1 libxss1
  python3-full python3-gdbm python3-tk python3.11-full tk8.6-blt2.5 whois x11-common
0 upgraded, 21 newly installed, 0 to remove and 0 not upgraded.
Need to get 16.8 MB of archives.
After this operation, 97.4 MB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 2to3 all 3.11.2-1 [10.0 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 libtcl8.6 arm64 8.6.13+dfsg-2 [973 kB]
Get:3 http://deb.debian.org/debian bookworm/main arm64 libxrender1 arm64 1:0.9.10-1.1 [32.0 kB]
Get:4 http://deb.debian.org/debian bookworm/main arm64 libxft2 arm64 2.3.6-1 [58.7 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 x11-common all 1:7.7+23 [252 kB]
[... 251 lines cut ...]
changed: [localhost]

TASK [system : Set cgroup_enable required by containerd for OOM] *****************************************************************************************************
changed: [localhost]

TASK [system : Set cgroup_memory required by containerd for OOM] *****************************************************************************************************
changed: [localhost]

TASK [system : Check if cdefs.h exists] ******************************************************************************************************************************
ok: [localhost]

TASK [system : Set fact for cdefs] ***********************************************************************************************************************************
ok: [localhost]

TASK [system : Remove libc6-dev] *************************************************************************************************************************************
changed: [localhost]

TASK [system : Install libc6-dev] ************************************************************************************************************************************
changed: [localhost]

TASK [system : Install Screenly dependencies] ************************************************************************************************************************
changed: [localhost]

TASK [system : Remove deprecated apt dependencies] *******************************************************************************************************************
6

nicomiguelino, 2024-05-03:
Thanks for providing results. Looks like a memory usage issue. Changes to the installation process can be made. Or maybe not use Docker, at least for Pi devices with smaller memory capacity.