Screenly/Anthias #1892: edits to cmdline.txt in wrong location

jdarmst, 2024-05-08:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

The ansible playbook edits /boot/cmdline.txt for several things. This is not longer the correct location for this file. It's at /boot/firmware/cmdline.txt.

#### Reproduction Steps

use install script on fresh install of raspbian lite 12 (bookworm)
check /boot/cmdline.txt

#### Environment

* Raspberry Pi Hardware Version: 3b+
* Raspberry Pi Network Setup: ethernet
* Anthias Version: latest as of today (5/8/2024)

nicomiguelino, 2024-05-08:
@jdarmst, thanks for noticing this. Editing the correct cmdline.txt could fix #1878.

nicomiguelino, 2024-05-14:
@jdarmst, please take a look at #1893 if you like to go into details. I'm still testing the changes though. #1878 is a known issue in devices running Bookworm. I hoe that the blinking top-left cursor goes away after the changes.

PR #1893 (2024-05-14): Sets `cmdline.txt` path conditionally
#### Overview

* Fixes #1892, #1878
* Set path to `/boot/cmdline.txt` for Bullseye and older
* Set path to `/boot/firmware/cmdline.txt` for Bookworm

#### Installing the changes

```bash
cd && \
echo "export CUSTOM_BRANCH='fix-1892'" > .env && \
bash <(curl -sL https://raw.githubusercontent.com/nicomiguelino/Anthias/custom-install-1/bin/install.sh)
```
closed 2024-05-14 by nicomiguelino