Screenly/Anthias #1914: video output is underscanned

tezza1971, 2024-06-05:
#### Overview of the Issue

The full output does not fit to the full hd display. It seems to be shrunken so that it is inset and there is a black border all around the output. I doubt this is intentional, so I will assime it's a bug. I am using a raaspberry pi 4B with the latest version of Anthias at the time of writing

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.2
* Raspberry Pi Network Setup: ethernet
* Anthias Version: master@aecc950
* Monitor Info: state 0xa [HDMI CUSTOM RGB lim 16:9], 1920x1080 @ 60.00Hz, progressive
* Memory 	Total: 3619 / Used: 663 / Free: 1973 / Shared: 31 / Buff: 38 / Available: 2874

nicomiguelino, 2024-06-05:
@tezza1971 By design, the display of video assets depends on the aspect ratio of the video. Will you be able to provide the following:

* Picture/snapshot of the issue
* The video file that you used. You can email me the video (or a link to it if it's too big) to nicomiguelino2014@gmail.com.

The display can have borders if, for instance, your display has 16:9 aspect ratio (e.g., 1920x1080) and the video used is not.

vpetersson, 2024-06-06:
@tezza1971 can you run `ffprobe` on the video in question?

tezza1971, 2024-06-06:
I understand that you can get black bars if the aspect ratio of the feed doesn't match the display, but this is a standard 1920x1080 test monitor with a standard 1920x1080 feed from a web URL. Even the splash screen and then the screen with the IP address is underscanned at bootup. I can't take a screenshot and i have no easy way to take a photo of the physical monitor and upload it here. It doesn't matter what content I specify. I use that monitor all the time with other computers and the the exact same rPi and I never have this issue. It's only with Anthias. Right now anthias is locked up. I dunno what's wrong with it. I will try and re-image the SD and start again.

tezza1971, 2024-06-06:
[image]
[image]
[image]

tezza1971, 2024-06-06:
I've encoded some h.264 video and that seems to work perfectly and uses the entire screen. Not sure why that is not the case for the rest of the playback. I uploaded a 1920x1080 jpg file and it was underscanned. I enabled all the "default" assets and all of them (screenly widgets) are underscanned resulting in the thick black space around the content.

In summary, only h.264 video content plays back normally on my Pi4B. Everything else is shrunken.

vpetersson, 2024-06-06:
Ah, there's a good explanation for this. Video playback and image/web playback uses different rendering methods. If you're seeing this only for images and web, then you likely need to tweak the overscan settings. This however can only be done currently if you're using the self-installed version on Raspberry Pi OS.

tezza1971, 2024-06-08:
I understand that you can get black bars if the aspect ratio of the feed doesn't match the display, but this is a standard 1920x1080 test monitor with a standard 1920x1080 feed from a web URL. Even the splash screen and then the screen with the IP address is underscanned at bootup. I can't take a screenshot and i have no easy way to take a photo of the physical monitor and upload it here. It doesn't matter what content I specify. I use that monitor all the time with other computers and the the exact same rPi and I never have this issue. It's only with Anthias. Right now anthias is locked up. I dunno what's wrong with it. I will try and re-image the SD and start again.

> Ah, there's a good explanation for this. Video playback and image/web playback uses different rendering methods. If you're seeing this only for images and web, then you likely need to tweak the [...]

Well I imagine that this adjustment would be included in the next Anthias image.

nicomiguelino, 2024-06-10:
@tezza1971, while I'm taking a look at the issue, I'd like to ask and confirm what OS you're using. Is it Raspberry OS Lite of Balena OS? If there's an Integrations menu/page in the web UI, then it's a Balena-based version of Anthias.

nicomiguelino, 2024-06-10:
@tezza1971, @vpetersson I already tweaked the device settings so that `disable_overscan` is set to `1`.
The fix should already be applied to your device by now.

closed 2024-06-10 by nicomiguelino

tezza1971, 2024-06-21:
Okay I will take a look to confirm. I haven't use this for a while. I will have to flash another ssd to test. Thanks for the update.