Screenly/Anthias #1928: Anthias stuck on Screenly screen

Stevemberg, 2024-06-13:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

- Format SD to NTFS or FAT32
- Use Imager or Balena (image from repository or downloaded) to make the recording
- Put the SD in the Raspberry pi
- Connect the network and video cables
- Turn on rasp
- Access the web page with the IP shown
- Upload an asset (image or URL)
- The screen with the Screenly logo does not change
[image]
[image]
[image]
[image]

#### Environment

* Raspberry Pi Hardware Version: RP 3B rev 1.2
* Raspberry Pi Network Setup: Ethernet
* Anthias Version: master@48e427b

nicomiguelino, 2024-06-13:
@Stevemberg
- How about adding another image asset?
- Have you tried adding a video asset or a web asset? Is the display stuck at the loading screen if its the only active asset?

Stevemberg, 2024-06-13:
- Yes, many images with differents formats and sizes.
- Yes, video don't change anything. Yes, same screen with many or one asset active

nicomiguelino, 2024-06-13:
@Stevemberg, I can see that the device name is not present. Since Anthias for Balena is a public fleet (see [Fleets for Good](`vc4-kms-v3d`) for details), fleet owners can access and monitor devices. Upon checking the device UUID, it seems that its non-existent on my end.

Stevemberg, 2024-06-13:
The device is not on the internet, just on an intranet.

Stevemberg, 2024-06-17:
I reinstalled one on another SD card and only turned it on when there was internet and then it got stuck on another splash screen.
[image]
I formatted the first SD and left it only on the intranet and it stuck on the same screen as the beginning of the issue.

nicomiguelino, 2024-06-17:
@Stevemberg, is your device that's connected to the Internet still stuck at the splash screen? For Raspberry Pi 3 devices, it will take a while for the resources to be finished downloading.

Stevemberg, 2024-06-18:
SOLVED! 
Leaving it connected to the internet for 20 minutes on the first boot solved the problem! Thanks for the tip that it would take a long time to download the resources.
Sorry for my inexperience, this is my first time working with embedded systems and IoT.

closed 2024-06-18 by nicomiguelino