Screenly/Anthias #1964: Request: Make Anthias work without internet

traeu, 2024-07-08:
#### Overview of the Issue

Related to Issue #1494, Issue #1773 and Issue #1949.

Some time ago, I noticed that Anthias won't run with internet access. After closer inspection, it became clear that TCP 443 access to api.github.com is needed. Without this access, the web interface of Anthias will become unstable and sometimes respond with nginx http error 504.
My workaround to this was to enable TCP 443 access to this server in our firewall.

After trying to update to the latest anthias-version, I noticed that this behaviour came back. Everything OK with internet-access allowed on TCP 443, but without, the web interface responds with error, and also Anthias doesn't show assets right anymore. I have assets switch every 10 seconds, but without internet access, the asset will only switch after about 3 to 5 minutes, not every 10 seconds.

I tried to log DNS requests and see what Anthias wants to connect to. At first sight, it looks like access to hub.docker.com is now also needed for stable operation. I enabled TCP 443 traffic to hub.docker.com in our firewall and so far, it looks like its working stable now. Of course I need to watch it for some days to see if it is really stable.

Can you confirm that right now, TCP 443 access to api.github.com and hub.docker.com is needed, and otherwise Anthias will have problems running stable?
Are there plans to remove this need for internet access?
Or maybe you can write in the installation documentation that access to certain external hosts is needed for stable operation and provide a up-to-date list of all external hosts that need to be accessible?

In Issue #1949, @nicomiguelino wrote that it should be possible to use Anthias on a network without internet access, and that web assets that are accessible internally, should be displayed without problems. Unfortunately, that's not working for me.

#### Reproduction Steps

Running latest Anthias without internet access.

nicomiguelino, 2024-07-08:
@traeu, I added a check (whether updates are available or not)  in the code where Docker tags are looked up from Docker Hub.

I can make changes so that it can work in an offline environment.

nicomiguelino, 2024-07-08:
@traeu, #1965 is still a work in progress. One limitation I'm facing is even though the problem seems to be fixed, the web UI pages now load slower, which is not good. I'll probably move the changes to a separate API endpoint and have JavaScript call it asynchronously.

vpetersson, 2024-07-09:
Just a note here - yes, it should be possible to use Anthias offline. *However* only if you're using a self-installed version and you set a local NTP server (since the Raspberry Pi doesn't have any RTC).

traeu, 2024-07-09:
Isn't that a common way to setup systems like this?
My setup:
Separate network without internet access. Gateway, DNS and NTP provided over DHCP. 
Firewall only allows access to the (internal) webservers that host the pages I want Anthias to display.
Access to internet only enabled temporarily when updates are made.
But because the Anthias-systems are in a separate network that can't be accessed from outside and that doesn't have internet access, it's not that important for me to make regular (security) updates.

vpetersson, 2024-07-09:
> Isn't that a common way to setup systems like this?

No, not really. Offline mode is pretty much a dying breed. Not even banks or other high security environments do it this way anymore.

nicomiguelino, 2024-07-09:
@vpetersson, good point. How about adding a setting in the web UI for enabling/disabling update checks?

vpetersson, 2024-07-09:
Feels unnecessary. We should just fail gracefully. Not being able to reach these end-points shouldn't break anything. The extent of it should be a "Unable to retrieve updates" message or similar.

closed 2024-07-10 by nicomiguelino

nicomiguelino, 2024-07-10:
@traeu, if you have some spare micro-SD card or Pi lying around, you can backup your data first before installing the updates. #1965 was merged recently.

traeu, 2024-07-15:
thanks for this update! I installed the latest version on all my Pis, now they are working like they should, even with disabled internet access and no exceptions for github and docker.
Perfect!

nicomiguelino, 2024-07-15:
@traeu, thanks for the confirming that you're not having the issue anymore. Glad to hear it!

argabarga, 2024-08-01:
I have a use case where I'm required to show content on a TV screen without any network access at all, meaning I need to be able to upload assets from disk. I couldn't find pointers on how to do that, but maybe someone in this thread would know?

I'm away from my raspberry at the moment, but it was installed around June 20th, so should be up to date at around that time.

vpetersson, 2024-08-01:
> I have a use case where I'm required to show content on a TV screen without any network access at all, meaning I need to be able to upload assets from disk. I couldn't find pointers on how to do [...]

At the very minimum, you then need to add a Real Time Clock (RTC) to the device if you have no network connection at all. If not, the device will have no clue about what time it is at all.

argabarga, 2024-08-01:
It would have to get the current time from its own internal clock, yes. Is that what I would achieve installing an RTC?

But like I said, the main obstacle for me was uploading assets.

vpetersson, 2024-08-01:
> It would have to get the current time from its own internal clock, yes. Is that what I would achieve installing an RTC?

Yes, but you can still be subject to time drift. 

> But like I said, the main obstacle for me was uploading assets.

You'd have to do a manual installation of Anthias on Raspberry Pi OS (with internet) and then configure the device to use a static IP and then wire up a device to it over Ethernet when you'd like to change content. Hardly ideal, but it's not a very common use case.

argabarga, 2024-08-01:
Hardly ideal, that's right. I'm allowed to bring disks with content, but not always allowed to bring a computer.

So maybe if I rephrase my question a bit: instead of having to upload, can Anthias be told to use assets on an external hard disk, connected via USB?

traeu, 2024-08-01:
Or you could use a Pi in your Lab/Office to prepare a SD-Card Image that contains Raspi OS, Anthias and all your content. Than write/clone this image to as many SD Cards as you need, bring them to this place and swap all sd cards of all your Pis there.
If there are just some images/videos in a loop, without the need for specific content at specific times, you maybe don't even need RTC because it would not matter what time the Pi thinks it is.

PR #1965 (2024-07-10): Fix: Making Anthias work without Internet
#### Description

* Fixes #1964 
* Modifies `is_up_to_date` so that it will fail gracefully.