Screenly/Anthias #1972: Webpage asset contents are not refreshed when it's the only asset active

rbauman70, 2024-07-13:
#### Overview of the Issue
Thanks for a great, easy-to-use product.  I have a Python Bottle application on the same Pi that is generating a listing of our events for the day.  The asset is defined as a URL that points to the server's webpage.  We also run a slideshow video as an asset. All works great as long as both assets are active.  Once the slideshow video asset is made inactive, the events listing asset no longer updates.  

From our events server log, I can see Anthias is making periodic requests, but once the HEAD request is made, it doesn't make a GET request on the page contents like it does when the other asset is active.

Here's the events server log showing when the slideshow video asset is inactivated and then reactivated.

Saturday, July 13, 2024 10:06 AM
172.18.0.3 - - [13/Jul/2024 10:06:20] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:06 AM
172.18.0.3 - - [13/Jul/2024 10:06:20] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:06:20] "GET /events.css HTTP/1.1" 304 0
172.18.0.3 - - [13/Jul/2024 10:06:20] "GET /favicon.ico HTTP/1.1" 404 748
Saturday, July 13, 2024 10:07 AM
172.18.0.3 - - [13/Jul/2024 10:07:10] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:07 AM
172.18.0.3 - - [13/Jul/2024 10:07:10] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:07:10] "GET /favicon.ico HTTP/1.1" 404 748
Saturday, July 13, 2024 10:07 AM
172.18.0.3 - - [13/Jul/2024 10:07:30] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:07 AM
172.18.0.3 - - [13/Jul/2024 10:07:30] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:07:30] "GET /favicon.ico HTTP/1.1" 404 748
Saturday, July 13, 2024 10:07 AM
172.18.0.3 - - [13/Jul/2024 10:07:50] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:08 AM
172.18.0.3 - - [13/Jul/2024 10:08:10] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:08 AM
172.18.0.3 - - [13/Jul/2024 10:08:30] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:08 AM
172.18.0.3 - - [13/Jul/2024 10:08:50] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:09 AM
172.18.0.3 - - [13/Jul/2024 10:09:10] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:09 AM
172.18.0.3 - - [13/Jul/2024 10:09:30] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:09 AM
172.18.0.3 - - [13/Jul/2024 10:09:50] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:10 AM
172.18.0.3 - - [13/Jul/2024 10:10:11] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:10 AM
172.18.0.3 - - [13/Jul/2024 10:10:11] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:10:11] "GET /events.css HTTP/1.1" 304 0
172.18.0.3 - - [13/Jul/2024 10:10:11] "GET /favicon.ico HTTP/1.1" 404 748
Saturday, July 13, 2024 10:11 AM
172.18.0.3 - - [13/Jul/2024 10:11:00] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:11 AM
172.18.0.3 - - [13/Jul/2024 10:11:00] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:11:01] "GET /favicon.ico HTTP/1.1" 404 748
Saturday, July 13, 2024 10:11 AM
172.18.0.3 - - [13/Jul/2024 10:11:50] "HEAD /events HTTP/1.1" 200 0
Saturday, July 13, 2024 10:11 AM
172.18.0.3 - - [13/Jul/2024 10:11:50] "GET /events HTTP/1.1" 200 1277
172.18.0.3 - - [13/Jul/2024 10:11:50] "GET /favicon.ico HTTP/1.1" 404 748

Here's the page's head section:
```
    <head>
        <title>CAB Events</title>
        <link rel="stylesheet" href="events.css">
		<meta http-equiv="Cache-Control" content="no-cache, no-store, must-revalidate" />
		<meta http-equiv="Pragma" content="no-cache" />
		<meta http-equiv="Expires" content="0" />
    </head>
```

#### Reproduction Steps

* Activate a dynamic content URL asset.
* Activate a local file video asset.
* Deactivate the local file video asset.
* Remaining dynamic asset content doesn't update.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.1
* Raspberry Pi Network Setup: wired Ethernet
* Anthias Version: master@dc1f963

vpetersson, 2024-07-13:
A HEAD request is done before showing an asset to make sure it's available. Since you seem to sending a 200 here, my guess would be that it's something in that logic that is broken. 

Ping @nicomiguelino

nicomiguelino, 2024-07-15:
@rbauman70, I revisited the code and it shows that a web asset does not refresh if it's the only active asset.
@vpetersson, unless it's by design, would it be sensible to add settings for toggling whether to refresh lone active assets?

nicomiguelino, 2024-08-08:
Duplicate of #1882

closed 2024-08-08 by nicomiguelino