Screenly/Anthias #1997: Can’t get Anthias working in Portrait 9:16 mode

nicomiguelino, 2024-07-27:


gdragotto, 2024-08-28:
Hello,

Any news on the matter? Can I help with the bug fix?

nicomiguelino, 2024-08-28:
@gdragotto, I haven't got back on this one. Open-source contributions are always welcome.
Meanwhile, you can check if adding `display_hdmi_rotate=1` (see [this documentation](https://www.raspberrypi.com/documentation/computers/legacy_config_txt.html#display_hdmi_rotate) for reference) will work. Don't forget to reboot your Raspberry Pi after modifying `/boot/firmware/config.txt` (for Bookworm) or `/boot/config.txt` (for Bullseye and older).

gdragotto, 2024-09-04:
Hi @nicomiguelino, thanks. It does not work unfortunately. I also played a bit with Anthias config, but still no luck from my side.

nicomiguelino, 2024-09-04:
[This comment](https://github.com/Screenly/Anthias/issues/1537#issuecomment-2329665242) from #1537 might help. You can also check if the following line is commented out:

```
dtoverlay=vc4-fkms-v3d
```

qbicdesign, 2024-10-07:
**Pi 3B, Raspberry pi OS lite, latest Athias as of today**

Commented out:
`dtoverlay=vc4-fkms-v3d`
added:
`display_hdmi_rotate=1`

**Result:**

- Display is rotated
- Anthias falls back to logo after initial white screen instead of displaying assets.
- After about 5 min it actually starts displaying.

Weird behaviour, but at least its working finally

nicomiguelino, 2024-10-07:
@qbicdesign,
* Did you reboot your device after modifying the config file?
* Can you [provide some logs](https://github.com/Screenly/Anthias/blob/master/docs/README.md#how-to-get-logs-from-anthias) during the time when Anthias falls back to the standby screen (the logo)?

qbicdesign, 2024-10-22:
1. yes
2. please be more specific about which logs would likely be of interest