Screenly/Anthias #2000: Missing install resources as of 7/29/24?

pr-wiley, 2024-07-29:
Today I am trying to install Anthias after creating a SD with pi imager.

The app seems to start up and a screen with the logo was presented. Then a white page noting a standard "Internal Server Error." I am guessing Anthias is looking for an on-line resource that not available.  After a bit the screen goes black and nothing seems to happen. 

Is a required web server down?

nicomiguelino, 2024-07-29:
@pr-wile, it will take a while for all the resources (including updates) to be downloaded. It might be that the viewer (which is responsible for controlling what content is to be displayed on the screen) displays a page from the server, which isn't available yet, maybe due to a download in progress.

I would like to check in and ask the following questions:
* What Raspberry Pi (hardware) version are you running Anthias on?
* Are you still getting the issue?

Let me flash my micro-SD card and see if the issue persists.

nicomiguelino, 2024-07-29:
@pr-wiley, I was able to set up my Pi 4 for Anthias (via the Raspberry Pi Imager). Let me know if you're still stuck in the same issue.

pr-wiley, 2024-07-30:
@nicomiguelino Thanks for the quick response.

Raspberry pi 4 B

I tried again this morning and the same sequence of events happened.  I'm going to leave the pi on all day and see what happens.

pr-wiley, 2024-07-31:
@nicomiguelino  OK, let the pi boot and run all day. Finally got to a dark blue screen with logo "Screenly" in the center of the screen with no change for several hours.

nicomiguelino, 2024-07-31:
@pr-wiley, it looks like the display is showing the old "Screenly" logo. We've changed the standby screen a couple weeks ago. (See screenshot below for reference.)

[image]

It was supposed to show the new loading page but it seems like its stuck at an older version. Can you try restarting/rebooting your device to see if a splash screen appears. The splash screen should show the IP addresses that can be accessed via the browser (for the web interface).

If, by any chance, you can get access to the web UI, can you go to the **_Integrations_** page and post the screenshot here?

nicomiguelino, 2024-10-30:
@pr-wiley, I just created new releases. Feel free to go to the releases page and download the image that fits your needs.