Screenly/Anthias #2003: Anthias doesn't work on Raspberry Pi 5

wpt-hsilva, 2024-07-31:
I have a raspi5 (64b) and I try to install Anthias with command 

$ bash <(curl -sL https://install-anthias.srly.io)

Everything seems OK, but when I restart the Raspberry after the installation process, an dark - grey window appears without any information.

I check the logs and I saw this
* journalctl -u docker
Jul 31 11:59:45 raspberrypi systemd[1]: docker.service: Start request repeated too quickly.
Jul 31 11:59:45 raspberrypi systemd[1]: docker.service: Failed with result 'signal'.
Jul 31 11:59:45 raspberrypi systemd[1]: Failed to start docker.service - Docker Application Container Engine.

*sudo systemctl restart docker
Job for docker.service failed because a fatal signal was delivered to the control process.
See "systemctl status docker.service" and "journalctl -xeu docker.service" for details.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 5 Model B Rev 1.0
* Raspberry Pi Network Setup: built-in WiF
* Anthias Version: latest

vpetersson, 2024-07-31:
This is expected. Raspberry Pi 5 is not supported yet.

closed 2024-07-31 by vpetersson

nicomiguelino, 2024-12-29:
@wpt-hsilva, we recently [added support for Raspberry Pi 5](https://github.com/Screenly/Anthias/pull/2174).
Feel free to install Anthias on your Raspberry Pi 5 device using any of the methods listed in [this documentation](https://github.com/Screenly/Anthias/blob/master/docs/installation-options.md).