Screenly/Anthias #2013: Settings page produces 504 Gateway Time-out

Pubec, 2024-08-07:
#### Overview of the Issue

Opening `settings` page produces **504 Gateway Time-out**.

#### Reproduction Steps

- installation by `$ bash <(curl -sL https://install-anthias.srly.io)`
- updated to latest version by ` bash bin/upgrade_containers.sh`
- open `http://<ip>/settings`

#### Environment

* Raspberry Pi 4 Model B Rev 1.2
* Ethernet connection, no WIFI
* Not using SSL
* Anthias version: master@6b8b0d0

nicomiguelino, 2024-08-07:
@Pubec, while investigation is in progress, I'd like to ask a few questions:

* What version of Raspberry Pi OS Lite are you using? (for instance, Buster or Bookworm)?
* Do you still access the homepage (**_Schedule Overview_**) and the **_System Info_** page?

nicomiguelino, 2024-08-07:
@Pubec, I tried to reproduced the issue but didn't replicated it. I answered `y` to all questions except the question that asks if I'd like to install the experimental version of Anthias. `master` is t `6b8b0d0`.

Pubec, 2024-08-07:
To answer the questions:

- What version of Raspberry Pi OS Lite are you using? (for instance, Buster or Bookworm)? **Bookworm**
- Do you still access the homepage (Schedule Overview) and the System Info page? **Yes**
- What's the memory capacity of your Pi 4 (for instance, 2 GB, 4 GB)? **4GB**
- Would you be able to check and provide the server logs, especially during the point that the issue occurred? _I've been check the compose logs during the request, but nothing even came op, or nothing unusual appeared_
- Right after running the installation script, you were prompted with four questions. I'd like to know your answers on those:
  - Do you still want to continue? (y/N) **Y**
  - Do you want Anthias to manage your network? This is recommended for most users because it adds features to manage your network. (Y/n) **Y**
  - Would you like to install the experimental version of Anthias instead? (y/N) **N**
  - Would you like to perform a full system upgrade as well? (y/N) **Y**

Logs:
- `docker logs screenly-anthias-nginx-1` produces **no** logs
- `docker logs screenly-antihas-server-1` produces:
```txt
[2024-08-07 07:38:30 +0200] [20] [INFO] Starting gunicorn 22.0.0
[2024-08-07 07:38:30 +0200] [20] [INFO] Listening at: http://0.0.0.0:8080 (20)
[2024-08-07 07:38:30 +0200] [20] [INFO] Using worker: gthread
[2024-08-07 07:38:30 +0200] [22] [INFO] Booting worker with pid: 22
host_agent_service is not ready after 60 retries
host_agent_service is not ready after 60 retries
host_agent_service is not ready after 60 retries
host_agent_service is not ready after 60 retries
```
- `docker logs screenly-anthias-websocker-1` produces **no** logs

nicomiguelino, 2024-08-07:
Thanks, @Pubec. With your logs, I already found the root cause: `host_agent_service is not ready after 60 retries` usually happens when retrieval of the device's IP address failed. I'll let you know when it's fixed.

In the meantime, I'd like to request specific logs from the system journal. You can do so by running `sudo journalctl -fu anthias-host-agent`, specifically during the time that the browser attempts to load the settings page.

Pubec, 2024-08-07:
Thanks @nicomiguelino . I've noticed the same error, and I've tried checking the `journalctl -fu anthias-host-agent`, but that provides me zero output, like there was nothing going on, even though anthias is running and working.

I can not get to `/splash-page`, but I can to main menu:
[image]

nicomiguelino, 2024-08-07:
@Pubec, I recently created #2014 in an attempt to fix the issue. I'll keep you posted. Do you get a white screen or something on `/splash-page`?

Pubec, 2024-08-07:
@nicomiguelino I get a 504 Gateway Time-out  aswell as at `/settings`

Could it possibly be a network manager issue, as I used `nmtui` to set fixed ip, with 23-bit netmask.

nicomiguelino, 2024-08-07:
@Pubec, I added a snippet in the settings view/controller that gets the IP.
I'm not sure if that's a Network Manager issue. Do you get similar issues if the IP addresses are not fixed?

Pubec, 2024-08-07:
@nicomiguelino should I checkout to to `nicomiguelino:fix-2013` branch and try to rebuild containers?

I can try removing connection from Network Manager with fixed IP.

nicomiguelino, 2024-08-07:
@Pubec, no not yet. I still have to have it approved and merged.
You can try disabling fixed IP first.

Pubec, 2024-08-07:
@nicomiguelino I've tried with fixed IP removal, and rebooted, but the problem persists. Only `/` and `/system-info` is accessible.

nicomiguelino, 2024-08-07:
@Pubec, I'll see what I can do. I an make some changes, just for you to test if out quickly as soon as possible, even if it's still not merged.
I'll give you the command later, just make sure to back up your data first before proceeding.

nicomiguelino, 2024-08-07:
@Pubec, once you've backed up your data, you can run the following to install a version of Anthias with my patches:

```bash
bash <(curl -sL https://raw.githubusercontent.com/nicomiguelino/Anthias/test-fix-2013/bin/install.sh)
```

Do take note that this is only temporary. If this went well, I'll have the pull request merge, then I'll let you know when changes to `master` is ready.

Pubec, 2024-08-07:
Whilst installing I did notice an error:
```
E: Conflicting values set for option Signed-By regarding source https://download.docker.com/linux/debian/ bookworm: /etc/apt/keyrings/docker.asc !=
E: The list of sources could not be read
```
Will get back with new updates as soon as docker pull finishes

Pubec, 2024-08-07:
Update finished, with error persisting: `docker logs screenly-anthias-server-1` produces `host_agent_service is not ready after 60 retries`

nicomiguelino, 2024-08-07:
@Pubec, you've said earlier that you got nothing when you ran `journalctl -fu anthias-host-agent`. It could be that Anthias is not properly and fully installed in your device.

You can also check if `anthias-host-agent.service` (or similar) is present inside the `/etc/systemd/system` directory.

If it's not too much on your end, can you do the installation from scratch and take note of the console output from start to finish (until the part where the script prompts you to reboot now or later). At that point, don't reboot your Pi yet, as you still need to get those console logs.

Pubec, 2024-08-08:
@nicomiguelino, unfortunately, there is nothing similar `in /etc/systemd/system` directory.

For a fresh install, should I reinstall Raspberry Pi OS as well?

nicomiguelino, 2024-08-08:
@Pubec, yes it's recommended to reinstall Raspberry Pi OS at this point.
Here's a screenshot of my terminal (that shows the contents of `/etc/systemd/system`) for reference:
[image]

nicomiguelino, 2024-08-09:
@Pubec. I'll be closing this for now. If the issue occurs again, please re-open this and include logs where the failure happened, for instance, in the installation script.

closed 2024-08-09 by nicomiguelino