Screenly/Anthias #2017: Movies are flickering and dont play smooth

Nezzu-bit, 2024-08-09:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

A paragraph or two about the issue you're experiencing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

* Press Add Asset
* Enter the Asset URL ...
* Press "On"
* Wait for the asset to show up on the screen

#### Environment

* Raspberry Pi Hardware Version:
* Raspberry Pi Network Setup: (ex. built-in WiFi, Ethernet, USB WiFi Adapter)
* Anthias Version:

nicomiguelino, 2024-08-09:
@Nezzu-bit, I'd like to ask a few questions:
* What version of Anthias is installed? (For example, `master@5474ff9`)
* If you are to check `/boot/firmware/config.txt` (Bookworm) or `/boot/config.txt` (Bullseye and older), what value is `dtoverlay` set to? &ndash; Is it `vc4-kms-v3d` or `vc4-fkms-v3d`? It would also help if you'd attach a copy of your `config.txt` here.

Nezzu-bit, 2024-08-12:
@nicomiguelino 

- The actual version is: experimental@84bd4ab
- config.txt ist under /boot/config.txt and the value of dtoverlay is set to vc4-fkms-v3d

config.txt is attached. :-)

nicomiguelino, 2024-08-12:
@Nezzu-bit, I just recently updated the experimental branch so that it will include the fixes to the flickering issue. The deployment is currently in progress. I''ll keep you posted.

nicomiguelino, 2024-08-12:
@Nezzu-bit, the updates were recently deployed. See this [CI run](https://github.com/Screenly/Anthias/actions/runs/10353128343) for referrence.

You can now install the updates, but make sure to back up your data first. Thanks.