Screenly/Anthias #2051: Black screen after each image

AchimPieters, 2024-09-04:
#### Overview of the Issue

After uploading, everything works, but between the images the screen turns black (flickers) and shows the next image? How to resolve this, so the images appear smoothly and disappear smoothly, without the black screen (and flickering) 

#### Reproduction Steps

Steps to reproduce this issue:
Uploaded two images of 1.150Kb with the size of 3840x2160px in .jpg format.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 4 Model B Rev 1.5 (8GB)
* Raspberry Pi Network Setup: Wi-Fi
* Anthias Version: v0.18.8 [Latest]

AchimPieters, 2024-09-04:
@nicomiguelino Thank you for your reply. Yes, it occurs also in other settings like 1920x1080px or 1280x720px (tested).

nicomiguelino, 2024-09-04:
Thanks, @AchimPieters. We're still assessing various solutions for the problem. We'll keep you posted. Changes are likely to be made on the C++ code. Open-source contributions are always welcome.

Slow-Learner-1, 2024-09-27:
I've had the same problem on a Pi 4 and 3B+ running Bookworm OS lite. I've found that if I convert my image files from .jpg to .png the transition between images is much smoother.

vseven, 2024-10-02:
Just wanted to comment that I'm having the same issue which seems to have been introduced in a recent version.  I have three RPi 4b's that were all running without issue.  Lost a SD card on one (stuck between wall and TV, assuming overheat) so replaced it and installed the latest version and having this issue.  Not very prominent between pictures (does happen sometimes) but happens often between picture and video to the point where the TV will sometimes think it's a new signal and pop up the "1920x1080" banner.  On the first play through it's usually fine, it's each play through afterward.  Our current rotation is only two slides then a video so tomorrow I'm going to test by making a much longer sequence and repeating those same 2+1 over and over to see if it's really ok the first play through and not the remainder.

AchimPieters, 2024-10-03:
For me, it's between small .mp4 files, as I don't have any other. Between each file, the screen goes black, no matter what I do or change. Maybe buffer related?

vseven, 2024-10-16:
> For me, it's between small .mp4 files, as I don't have any other. Between each file, the screen goes black, no matter what I do or change. Maybe buffer related?

Yeah, the black screen issue is back for me.  Nothing changed, just came back a couple days later and is popping up the TV's signal thing.  Not sure why no one else is having the issue unless they just don't use mp4's.

maedangad, 2024-10-24:
Nice to meet you, may I ask you a question?
When I play a video with Anthias, a blackout of about two seconds occurs during video transitions.
Do you know any settings or equipment I could use to prevent this issue?

nicomiguelino, 2024-10-25:
We're still assessing options that could be a drop-in replacement for the video player that Anthias is currently using right now, which is VLC.

One of the option is to transition to Qt's multimedia player, but we still have to make sure that the change will not affect the devices (for all the supported platforms).

liquido001, 2024-11-14:
Is there any changes solving this issue?

nicomiguelino, 2024-11-14:
@liquido001, there aren't any changes yet, but feel free to contribute and open a pull request.

AchimPieters, 2025-01-11:
@nicomiguelino Yesterday, I updated Anthias to the latest version, and I noticed that the transition is much smoother and faster now! Instead of a black screen, there’s now a swipe-to-black effect, and the next MP4 is displayed immediately. This is definitely an improvement!

bhagemann, 2025-01-14:
Worth a try, so I just ran a new install with latest Anthias. I still get the pause after the first inch of slide shows on screen, then the rest displays for the time i have set, then a black 2sec pause between slides, repeat.  RaspPi3 with Raspberry Pi OS Lite (64bit) installed using Raspberry Pi Imager on new 32gig sd card, installed Anthias from ssh terminal.  On wifi, uploaded 4 1920x1080 jpgs to rotate between.  no other mods to settings.  TIA for any insight.

AchimPieters, 2025-01-23:
@nicomiguelino Yesterday, I saw that there was an update available and run it. Unfortunately, the problem is back!

PJF16, 2025-03-26:
I have the same problem even with a Raspberry Pi5 8GB.

nicomiguelino, 2025-03-29:
I'll be opening another pull request that will actually resolve this issue.

nicomiguelino, 2025-04-02:
@PJF16, if you installed Anthias on a Raspberry Pi 5 running 64-bit Raspberry Pi OS Lite, you can try updating by running the install script again.

@AchimPieters, deployment of fixes for Raspberry Pi 4 devices is still ongoing.

bhagemann, 2025-04-02:
anything for the Pi3?  I think I have a spare 4 I can repurpose, but our 2 in use are Pi3's which I reverted to a older Screenly that is working fine for slideshow use.

nicomiguelino, 2025-04-03:
@bhagemann, I just tested it earlier on my Pi 3 (Model B+) and transitions are now seamless, at least between images.
Let me know if you're still having the same issues.

PR #2253 (2025-03-29): fix: black screen after each image
### Issues Fixed

- Fixes #2051

### Description

- This is an attempt to avoid a gap (in a form of a black screen) between assets.

### Checklist

- [x] I have performed a self-review of my own code.
- [x] New and existing unit tests pass locally and on CI with my changes.
- [x] I have done an end-to-end test for Raspberry Pi devices.
- [x] I have tested my changes for x86 devices.
- [x] I added a documentation for the changes I have made (when necessary).
PR #2254 (2025-04-02): fix: only display a web asset when it has finished loading
### Issues Fixed

#2253 fixed #2051, but let's say you have the following playlist items (in order):

- Image Asset 1
- Web Asset 1
- Image Asset 2
- Web Asset 2

When Web Asset 2 is about to be displayed, Web Asset 1 can still be seen for a quick moment.

### Description

Waits for a web asset to be finished (loading) before showing it.

### Checklist

- [x] I have performed a self-review of my own code.
- [x] New and existing unit tests pass locally and on CI with my changes.
- [x] I have done an end-to-end test for Raspberry Pi devi [... cut]
PR #2255 (2025-04-02): fix: update WebView version and hash
### Issues Fixed

- Fixes #2051

### Description

- Updated WebView version used by Anthias to `v0.3.7`.

### Checklist

- [x] I have performed a self-review of my own code.
- [x] New and existing unit tests pass locally and on CI with my changes.
- [x] I have done an end-to-end test for Raspberry Pi devices.
- [x] I have tested my changes for x86 devices.
- [x] I added a documentation for the changes I have made (when necessary).
closed 2025-04-03 by nicomiguelino