Screenly/Anthias #2066: No soundcards found

ChrisiHill, 2024-09-11:
#### Overview of the Issue

Hi, I have installed Anthias via bash <(curl -sL https://install-anthias.srly.io) on raspberry pi os lite. since installing I have lost my soundcards.

aplay -l
aplay: device_list:274: no soundcards found...

cat /proc/asound/cards
 0 [vc4hdmi        ]: vc4-hdmi - vc4-hdmi
                      vc4-hdmi
 1 [Headphones     ]: bcm2835_headpho - bcm2835 Headphones
                      bcm2835 Headphones

I had pulseaudio working via hdmi before, and was able to get into the raspi-config audio menu, but now if I click on that it immediately brings me back to the system menu.

This is my second time installing Anthias with the same issue.

I am trying to achieve audio only music via vlc over hdmi to the tv, and this for video. I had it briefly work yesterday before I ran the upgrade script, which broke it, but even a fresh install get it working.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi 3 Model B Rev 1.2
* Raspberry Pi Network Setup: built-in WiF
* Anthias Version: master@8d74226

ChrisiHill, 2024-09-11:
Also just to add, I was also able to stream music via vlc before i installed, after it would appear to play, but nothing would be coming from the speakers

nicomiguelino, 2024-09-11:
@ChrisiHill, I'd like to ask a few questions:
* What OS version runs in your Raspberry Pi 3? (e.g., Bullseye, Bookworm)
* Is video playback on your Pi still working well? (You've mentioned earlier that you're trying to achieve audio-only music.)
* Have you tried going to the **_Settings_** page in the web UI then selecting an option from the **_Audio output_** drop-down?

[image]

ChrisiHill, 2024-09-11:
Hi @nicomiguelino, Thanks for the quick response.

It was a fresh install of Bookworm 64 bit.

I did try video playback via pasting in a youtube URL when uploading an asset, but the audio of that did not play.

I did try changing it to both hdmi and 3.5 jack, just to see if the hdmi audio was already in use.

nicomiguelino, 2024-09-12:
Have you tried uploading a video file from your computer (not via a YouTube link) and see if there's audio?

ChrisiHill, 2024-09-12:
How do I rollback to the previous version, as it was fine the first time I installed it, then broke after I updated, and now with a fresh install it just installed the latest version.

ChrisiHill, 2024-09-12:
Pulseaudio was the issue. Using pipewire now.

closed 2024-09-12 by ChrisiHill