Screenly/Anthias #2070: Failed to create secure directory (/data/.config/pulse): Operation not permitted

Mattewoosh, 2024-09-13:
Hi, after building balena cloud release and installing balenaOS on Raspberry Pi 4, or when using already builded image, I get following errors while playing .mp4 files or giving website urls (website.com/video.mp4):

[image]

video itself plays just fine, but there isn't any audio.

nicomiguelino, 2024-09-13:
@Mattewoosh, while I'm taking a closer look, I'd like to ask for the following details:

* Are you using a pre-built image (which runs on BalenaOS)?
* What version is your device currently in? (e.g., `master@bb31a1b`)
  * You can check the version by going to the web UI going to the **_Settings_** page. (See **_Anthias Version_**.)
* What's the UUID of your device?
    * You can go to the **_Integrations_** page and find the **_Device UUID_**.

Mattewoosh, 2024-09-13:
1. I tried building the lastest release, pre-built image, and now it's running on v0.18.7. Same problems on all of those.
2. master@8d74226
3. 3e690a5a3548f1e72907045101418d9b

nicomiguelino, 2024-09-13:
@Mattewoosh, have you tried any of the following?

* Change the Audio output from HDMI to 3.5mm jack via the Settings page?
* Connecting to the display via **_HDMI0_** instead of **_HDMI1_**?
  * Try to restart your device if you're changing the HDMI connection.

nicomiguelino, 2024-09-13:
I can't seem to access the device using your device ID. I can now see that you've deployed Anthias to your own fleet.

Have you tried configuring your device via the following commands (from your host machine)?

```bash
$ balena env add BALENA_HOST_CONFIG_gpu_mem $GPU_MEM_VALUE --fleet $FLEET_NAME
$ balena env add BALENA_HOST_CONFIG_dtoverlay vc4-kms-v3d --fleet $FLEET_NAME
```

Just substitute `$GPU_MEM_VALUE` and `$FLEET_NAME` with the desired value.

It was mentioned in [this documentation](https://github.com/Screenly/Anthias/blob/master/docs/balena-fleet-deployment.md) that `BALENA_HOST_CONFIG_dtoverlay` should be set to `vc4-fkms-v3d`, which is outdated. It should be `vc4-kms-v3d`. I'll make changes to the docs. Please let me know if that fixes the issue.

Mattewoosh, 2024-09-14:
Hi, changing BALENA_HOST_CONFIG_dtoverlay to vc4-kms-v3d solved the issue, thank you for help.

closed 2024-09-14 by Mattewoosh