Screenly/Anthias #2090: Wifi connect isn't working, goes to white failure screen

Planks, 2024-10-11:
#### Overview of the Issue

Don't seem to be able to connect to the pi with wifi in any way

#### Reproduction Steps

I've tried installing the latest built to both a raspberry 3 and 4 by a few methods:

Pi imager (bundle)
Pi imager (download from release page)
Raspbian lite install script

Only by going through raspbian have I been succesful at obtaining wifi access, using the native controller. On both the Anthias builds it goes to a white page, which I assume is supposed to be the wifi access point connection thing (I used Screenly OSE a lot, I know that process), before falling over onto a logo and black screen

Via raspbian lite and the pi 3 I had a different problem, the system lagged a lot (1 frame per minute video, 1920x1980 image took 3 seconds to load), so I gave up on that one and don't care to try to reproduce it due to the install being particularly time consuming

There's no wired internet here to connect to - am I doing something wrong or is this build broken?

#### Environment

* Raspberry Pi Hardware Version: 3, 4
* Raspberry Pi Network Setup: built in wifi
* Anthias Version: 0.18.9

[image]

nicomiguelino, 2024-10-11:
@Planks, the `wifi-connect` component is not working at the moment, but feel free to open a pull request (if possible). Otherwise, I'll keep you posted  about the fixes.

nicomiguelino, 2024-10-11:
@Planks, for BalenaOS-based images, it's probably not possible.
If you're trying to flash a micro-SD card with RPi OS, you can configure Wi-Fi settings in the meantime.

vpetersson, 2026-05-08:
The `wifi-connect` captive portal was removed entirely in #2763 (merged 2026-04-28) — the current flow expects you to set up Wi-Fi via Raspberry Pi Imager or `nmcli`/`nmtui`. Closing.