Screenly/Anthias #2108: Raspberry Pi Zero 2 W Support?

mcrosson, 2024-11-03:
#### Overview of the Issue

Are there any plans to officially support the Raspberry Pi Zero 2 W?

#### Reproduction Steps / Why I Ask

I ask about official support because I've been experimenting with running Anthias on the Pi Zero 2 W and I keep running into horrific disk io contention issues. An overview of my progress is below.

I was able to sort the installer problems I found and was able to get the install to run to completion within 45-60 minutes on average. However, during container start up the disk io contention makes the Pi Zero 2 W unusable. I noticed something similar on my pi4 deployment: whenever the containers start, they use up all the disk io, cause the load average to spike over 15-20 and the system becomes unusable. On the pi4 the heavy disk io eventually settles and the load average drops within a 10 minutes usually but these two things can be very problematic during startup on the pi4.

On the Pi Zero 2... I've yet to find a way to constrain the intense disk io the containers cause during startup to allow Anthias to actually run. I'm seeing load averages as high as 60 when starting the containers without tuning and with some significant tuning I'm seeing load averages over 30. In either scenario the system becomes unusable very quickly. Additionally, the system will eventually throw a kernel error saying a task is hung. Once this happens, the Pi Zero 2 W is effectively crashed and unusable. The server and viewer containers seem to cause the biggest problems for disk io contention in my experiments.

From what I can figure out, the containers would need to be started sequentially and not in parallel and/or tuned to be less aggressive on disk io during startup. Without the disk io contention addressed, I don't see how the Pi Zero 2 W could be supported.

I can provide my notes if desired but they are dense and very long with a pair of patches that make some significant changes to the setup process and docker-compose template. Please let me know if you'd like this additional detail.

#### Environment

* Raspberry Pi Hardware Version: Raspberry Pi Zero 2 W
* Raspberry Pi Network Setup: Built in WiFi
* Anthias Version: GitHub sources, revision 8acb12d683250c7a9b9fa21590d62ae4e98056fb

nicomiguelino, 2024-11-04:
@mcrosson, thank you for creating an issue. Yes, I'd be happy to see your notes. I don't mind if they're dense. I really appreciate it.

mcrosson, 2024-11-04:
## reply to @nicomiguelino 

the below is everything ive found related to the raspberry pi zero 2 w. a lot of the changes will likely need adjusting if incorporated into the main anthias project. you're welcome to borrow/use/etc anything from the below.

i hope the below has enough clarity to be useful but if not, i can provide any additional detail as desired.

## overview of issue

the anthias installer hangs and sometimes crashes when run on a raspberry pi zero 2 w -- its essentially 'impossible' to install anthias on a pi zero 2 w without changes to the anthias install procedure. 

i managed to find fixes that allow anthias to be installed on a pi zero 2 w. these fixes are fully documented below. 

i also found additional items that should be addressed for better pi zero 2 w support. these items are also documented below. some of these may also be applicable to other raspberry pi setups.

i don't know how the maintainers may want to proceed and have placed my notes here for reference. hopefully they will allow for another person to put together patch(es) based on my notes that meet the requirements for inclusion in the project.

## environment / hardware

- Hardware Version: Raspberry Pi Zero 2 W
- Networking: Built in WiFi on 2.4ghz with dhcp
- Code: Latest github sources
- OS: Raspberry Pi OS Lite (64bit)
- Storage: multiple different sd cards, multiple usb-otg disks

## related github issues (probably)

- https://github.com/Screenly/Anthias/issues/2055
- https://github.com/Screenly/Anthias/issues/1962

## critical / must read before continuing

the below *will* allow anthias to be installed on a pi zero 2 w with an sd card.

**HOWEVER: IT WILL NOT RUN PROPERLY POST INSTALL**

post install, anthias will cause load average spikes over 40 and will generally Not Work at all.

anthias *requires* a storage setup thats fast enough to keep pace with the containers. the containers perform heavy disk io and will make the setup wholly unusable on the pi zero 2 w. i have tested multiple sd cards and usb-otg disks, all *fail* to run the anthias services post-install.

i cannot underscore this point enough: anthias will **not run** from an sd card or usb-otg properly on the pi zero 2 w based on my testing.

## fundamental fixes for installer

### swap

swap is *required* for the install to work correctly. the current installer removes `dphys-swapfile` *and* disables swap mid-way through the install process. disabling swap via these steps causes problems with disk io to the point the device becomes unusable and the install will eventually fail or lock up so badly (due to high load average) the installer may as well have failed to run.

keeping `dphys-swapfile.service` and swap enabled for the duration of the install avoids disk io stalls, keeps ssh over wifi interactive and greatly reduces time to install anthias. this also solves all pi zero 2 w install issues ive seen reported on github related to 'hangs' or related to apt failing (particularly on libc6-dev and docker install).

ive included a work around / fix below.

### docker pull

docker tries to download and extract images in parallel which causes high load averages and lock ups. setting the `max-concurrent-downloads` option to `1` will generally avoid lock ups related to `docker pull`. this is required for the install to complete. 

ive included a work around / fix below.

## additional concerns (and fixes) beyond the installer

these items are mainly related to *running* anthias. the below will make anthias *seem* to start but in reality it will ultimately fail to run due to heavy disk io which causes load average spikes. the below changes mitigate this to an extend but to *not* fully solve the problem.

note: ive left the system running overnight and it *still* had failed to start the anthias containers. the below mitigates problems running anthias but does not fully solve them.

- the cgroup config that is needed to allow docker to limit container memory use is not setup by default
	- the cgroup memory config needs to be set to prevent the containers from eating all the boards ram
	- ive included a work around / fix for this below
- during the cleanup phase of the install script...
	- the system becomes unusable during this phase of installation due to the containers being started prior to the install completing 
	- ive included a work around / fix below
- cpu spikes caused by containers can be an issue
[... 463 lines cut ...]
- with dphys-swapfile and swap active the device remains interactive over wifi
	- heavy disk io operations make working in other tmux windows problematic
		- apt operations
		- docker container pull operations
	- best to stick to one tmux window & no other work being performed while the install is active
	- pane for install and pane for htop seems to be ok
- apt operations heavily use the swap file
- docker install step eats the *entire* swap file
- the disk cache/buffer in memory is *large* for duration of run -- the swap file keeps io write pressure low by swapping background services and using ram for disk io buffering from what i can tell in my testing
- heavy disk iops cause load average spikes due to storage being slow on write
	- if load average gets to >=8, terminal output over wifi will almost always stop and/or hang
- wifi stalls at times and can be very problematic for interactive terminals
	- use tmux or screen to be safe
[... cut ...]
- watched io utilization with htop during troubleshooting and fix development
- hangs reliably on libc6-dev remove task with swap disabled
	- this causes >= 8 load averages
- hung reliably on docker install steps with swap disabled
	- this causes >= 8 load averages
- hung reliably on docker pull with swap disabled
	- this causes >= 8 load averages
- hung at other points with swap disabled, every time is >= 8 load average
- the ansible apt steps, docker install and docker pull seem to be the biggest cause of problems with swap disabled

nicomiguelino, 2024-11-04:
@mcrosson, thank you so much! I'll spend a fair bit of time looking into those details. I'll let you know if I created a pull request based from the patches that you've provided.

nicomiguelino, 2024-11-04:
@mcrosson take note that as an option, you can also create a pull request if you'd prefer. (We appreciate open-source contributions from users.) I'll just drop comments and suggest changes if needed.

mcrosson, 2024-11-04:
> @mcrosson take note that as an option, you can also create a pull request if you'd prefer. (We appreciate open-source contributions from users.) I'll just drop comments and suggest changes if needed.

I'll definitely keep this in mind in the future. I think some of the above would make sense 'in general' but until I get hands on additional test hardware, I think it's wise to not post any of the above as PR's myself.

mcrosson, 2024-11-04:
> @mcrosson, thank you so much! I'll spend a fair bit of time looking into those details. I'll let you know if I created a pull request based from the patches that you've provided.

Sounds good. I'll have a Pi Zero 2 W on-hand for testing/info gathering. Just let me know if you need anything more from me.

I'll also keep an eye on this ticket so I don't miss any additional comments or feedback.

nicomiguelino, 2024-11-04:
@mcrosson, will take note of that. For now, I'll test the patches based off the notes, and then create PR. Thanks!

nicomiguelino, 2024-11-04:
@mcrosson, I created a PR from your patches. Let me know if I missed anything from the patches. Some of them were rejected, so I have to `git apply` some of the changes manually. I will still have to test them.

mcrosson, 2024-11-04:
the only thing i see missing is `dphys-swapfile` being removed from `ansible/roles/system/tasks/main.yml` -- `dphys-swapfile` needs to be present on the system and running during the anthias install process or the install will fail.

mcrosson, 2024-11-04:
I just realized you could probably set the restart policy on all the containers to `never` and keep `anthias-host-agent.service` disabled during testing/development.

Those two adjustments should prevent the containers from auto-starting on boot so if the board crashes during testing you can simply remove power to turn off the board and then cold boot to resume where you left off.