Screenly/Anthias #2116: cannot activate items created with the api via the web gui

mcrosson, 2024-11-06:
#### Overview of the Issue

When using the api to create new assets in anthias, I cannot use the web gui or the api to activate the files. I can enable/disable but activation does not seem to work.

I ran the Firefox inspection tools and as far as I can tell my requests to the api endpoint are valid but they have no effect.

The code I've put together is [here](https://gist.github.com/mcrosson/8b4b4a0740ff55394fb78a4ac0095ffd).

Any help/insights with using the api to add/enable/activate files would be appreciated. I cannot figure out what I'm missing.

#### Reproduction Steps

Steps to reproduce this issue, eg:

Add http assets using the v1.2 api endpoints using urllib3 and try to enable them via the main web ui

#### Environment

* Raspberry Pi Hardware Version: raspberry pi 4
* Raspberry Pi Network Setup: built in wifi
* Anthias Version: latest container updates as of 2024-11-06

mcrosson, 2024-11-07:
It looks like you need a positive delta between start and end time when creating an asset. I was not aware of this and using a positive time delta between start and end time seems to have worked.

I somehow missed this when exploring the api via the Firefox inspector.

nicomiguelino, 2024-11-07:
@mcrosson, apologies for a late reply. I'm glad that you've figured it out.
* Yes, you have to make sure that the time difference is always positive.
* You also have to make sure that the datetime instances are in UTC since date and time info saved in Anthias are in UTC as well. For example:
```python
from datetime import datetime, timezone

start_date = datetime.now(timezone.utc)
```

Feel free to create a pull request with API documentation changes if you prefer. Otherwise, I'll just add it to the project board.

You can also check the API docs in your browser via `http://<IP_ADDRESS>/api/docs`.