Screenly/Anthias #2121: Building WebView Question

mcrosson, 2024-11-10:
I'm working on building Anthias for [Armbian](https://www.armbian.com/) and quickly discovered the `WebView` part of the Anthias project needs to be built from source. I also noticed the current `WebView`  arm build process is setup very specifically for the Raspberry Pi boards.

During my research I discovered there are 2 open pull requests, #2081 and #1888 , that seem to make the necessary changes I'd want to be able to build `WebView` for Armbian. They both seem to accomplish the same thing but in different ways.

From what I can tell #1888 should be able to build the `WebView` for qt6 and have it work with the viewer while #2081 has a note that the compile completes but the output doesn't work in the viewer. 

Is the above an accurate read of the PR details? If so, would it be best to work from #1888 or try #2081 first and then fall back to #1888 or should I focus on the current qt5 build process already present while the two open PRs are worked on by the main Anthias devs?

It's not clear to me which approach of the above would be the most appropriate choice and I'd appreciate any guidance that can be provided.

nicomiguelino, 2024-11-10:
@mcrosson, I'm still assessing options on how to cross-compile Qt 6.
* #2081 makes use of pre-build Qt 6 binaries installed via `apt`. Yes, I tried to compile it for ARMv7 but doesn't work when I run it on a containerized Raspberry Pi 3.
* Installing pre-built Qt 6 binaries via `apt` might have some limitations. It might be a better option to build Qt from source.
* Feel free to share your changes  just in case you came up with fixes/solutions.

mcrosson, 2024-11-10:
Is there a reason to have multi-stage builds with cross compile for arm builds instead of a set of single stage builds leveraging the `--platform` flag? Multi-platform is already a hard requirement for building WebView and it seems like single stage builds would be possible and avoid some of the complexities of cross compilation.

nicomiguelino, 2024-11-11:
@mcrosson, Multi-stage builds is needed in cross-compiling Qt, since the Raspberry Pi sysroot (from the first stage) is needed.