Screenly/Anthias #2133: Cannot install due to "sudo: a password is required" error Debian Bookworm (x86)

shaqaruden, 2024-11-18:
When filing a bug, please include the following headings if possible. Any example text in this template can be deleted.

#### Overview of the Issue

When attempting to run the installer with the following options selected:
Managed Network: No
Full System Update: No
Branch: master or v0.19.1

The installation fails at TASK [Gathering Facts] with the following error message
[image]

#### Reproduction Steps

Steps to reproduce this issue, eg:

1. Install Debian Bookworm
2. Update OS `sudo apt update && sudo apt upgrade`
3. Run install command from user directory `bash <(curl -sL https://install-anthias.srly.io)`

#### Environment

* OS: Debian Bookworm
* Arch: x86/x64
* Anthias Version: master or 0.19.1

vpetersson, 2024-11-18:
@nicomiguelino There's an ansible-playbook argument for ansible to ask for password that needs to be added iirc.

nicomiguelino, 2024-11-18:
@vpetersson, thanks. I think it's the '--ask-become-pass', as seen in this [documentation](https://docs.ansible.com/ansible/latest/playbook_guide/playbooks_privilege_escalation.html#become-command-line-options). Let me create a pull request for this.

shaqaruden, 2024-11-18:
> @vpetersson, @shaqaruden Let me create a pull request for this.

Thanks!

nicomiguelino, 2024-11-18:
I just created a pull request, #2134. Let me just test the changes.