Screenly/Anthias #2158: Is Anthias on Raspberrypi 5 available

PiBuilder5, 2024-12-11:
Hi Guys,
i have tryed to install Anthias on my RaspberryPi 5, but it didn´t works.

#### Overview of the Issue

Installed from github on RaspberryPi 5, but the script does not complete the installation.

#### Environment

* Raspberry Pi Hardware Version: Raspberrypi 5
* Raspberry Pi Network Setup: Ethernet
* Anthias Version:

nicomiguelino, 2024-12-11:
@PiBuilder5, support for Raspberry Pi 5 is not yet available. We'll keep you posted once it's ready.
Also feel free to create a pull request if you have some changes that you would like to be merged.

nicomiguelino, 2024-12-16:
@nstephens, I'm still working on that one. Thanks for sharing. The installer (from `master`) installs the `armhf` version of Docker. You can try installing  the `arm64` version instead:

```
    # Add Docker's official GPG key:
    sudo apt-get update
    sudo apt-get install ca-certificates curl
    sudo install -m 0755 -d /etc/apt/keyrings
    sudo curl -fsSL https://download.docker.com/linux/debian/gpg -o /etc/apt/keyrings/docker.asc
    sudo chmod a+r /etc/apt/keyrings/docker.asc

    # Add the repository to Apt sources:
    echo \
    "deb [arch=$(dpkg --print-architecture) signed-by=/etc/apt/keyrings/docker.asc] https://download.docker.com/linux/debian \
    $(. /etc/os-release && echo "$VERSION_CODENAME") stable" | \
    sudo tee /etc/apt/sources.list.d/docker.list > /dev/null
    sudo apt-get update

    # Install Docker:
    sudo apt-get install -y \
        docker-ce \
        docker-ce-cli \
        containerd.io \
        docker-buildx-plugin \
        docker-compose-plugin

    # Add the current user to the docker group:
    sudo usermod -aG docker $USER
```

You can also check the following documentation for details:
* https://docs.docker.com/engine/install/debian/
* https://docs.docker.com/engine/install/raspberry-pi-os/

nstephens, 2024-12-16:
thanks @nicomiguelino ! I used the above and was able to successfully update docker to a supported version. I then manually ran the upgrade_containers.sh script which worked to some extent at least:

```
$ ./screenly/bin/upgrade_containers.sh 
fatal: not a git repository (or any of the parent directories): .git
[+] Pulling 70/7
 ✔ anthias-server Pulled                                                                                    378.3s        
 ✔ redis Pulled                                                                                             406.9s        
 ✔ anthias-celery Pulled                                                                                    526.0s        
 ✔ anthias-websocket Pulled                                                                                 515.1s        
 ✔ anthias-wifi-connect Pulled                                                                              514.9s        
 ✔ anthias-nginx Pulled                                                                                     390.8s        
 ✔ anthias-viewer Pulled                                                                                    535.8s        
[+] Running 9/10
 ✔ Network screenly_default                   Created                                                         0.2s        
 ✔ Volume "screenly_redis-data"               Created                                                         0.0s        
 ✔ Volume "screenly_resin-data"               Created                                                         0.0s        
 ✔ Container screenly-redis-1                 Started                                                         9.1s        
 ⠇ Container screenly-anthias-server-1        Starting                                                        9.1s        
 ✔ Container screenly-anthias-viewer-1        Created                                                         0.2s        
 ✔ Container screenly-anthias-websocket-1     Created                                                         0.2s        
 ✔ Container screenly-anthias-celery-1        Created                                                         0.2s        
 ✔ Container screenly-anthias-nginx-1         Created                                                         0.2s        
 ✔ Container screenly-anthias-wifi-connect-1  Created                                                         0.2s        
Error response from daemon: error gathering device information while adding custom device "/dev/vchiq": no such file or directory
```

Bush-cat, 2024-12-17:
The /dev/vchiq interface was deprecated on the pi 5 as it was more of a temporary direct access to the gpu.
It seems Anthias needs a few more changes to work on the pi 5.

nicomiguelino, 2024-12-17:
@Bush-cat, thanks for the feedback. Yes, I agree. We also need to push Docker images compatible with the Raspberry Pi 5.
We're currently working on that in #1868. While the PR is still a work in progress (and in draft), feel free to give feedback.

closed 2024-12-20 by nicomiguelino

nicomiguelino, 2024-12-20:
@nstephens, @Bush-cat, @PiBuilder5 We just shipped the changes for Raspberry Pi 5. Feel free to run the install/update script.

PR #1868 (2024-12-20): Adds support for Raspberry Pi 5
### Overview

- Adds support for Raspberry Pi 5
- Fixes #2158

### Checklist

- [x] Add Pi 5 support for installing Anthias on Raspberry Pi OS Lite.
- [x] Add Pi 5 support for devices running **_balenaOS_**.
- [x] Update download link for Qt and WebView binaries.

### How to test the changes?

```bash
cd && \
bash <(curl -sL https://raw.githubusercontent.com/nicomiguelino/Anthias/refs/heads/test-rpi5-support/bin/install.sh)
```