Screenly/Anthias #2165: Update the build workflow for Docker images so that images per service are build concurrently

nicomiguelino, 2024-12-18:
### Overview

* Builds are currently divided by platform (Pi 1, 2, 3, 4 & x86).
* The build process can further be optimized by splitting image builds (per service) into separate jobs (via matrices), where service $$\epsilon$$ {server, celery, redis, websocket, nginx, viewer, wifi-connect}.
* This means that there will be a total of 5 x 7 = **35** concurrent image build jobs running, in theory.

vpetersson, 2024-12-18:
Yes, but i think we would gain a lot more by just optimizing caching. We could do both of these in parallel of course, but it should be highly cacheable.

nicomiguelino, 2025-04-17:
This was already fixed by #2267.
Caching-related fixes will be addressed in a separate pull request.