Screenly/Anthias #2190: [BUG] create asset api returns 500 error but still creates asset

jay1459, 2025-01-03:
**Describe the bug**

Using the create asset api, a 500 error is returned

Response data: {
  error: "'<' not supported between instances of 'str' and 'datetime.datetime'"
}
Response status: 500

**To Reproduce**

Steps to reproduce the behavior:
1. create an assetfile using the asset file api "/api/v2/file_asset"
2. use postman to create the asset using the create asset api "/api/v2/assets"
3. mine had the following request data

Request data: {"ext":".jpg","name":"WP_20140120_18_31_24_Pro.jpg","uri":"/data/screenly_assets/845bb3ebc091504283ab189c40b4a416.tmp","start_date":"2025-01-03T09:57:48.753Z","end_date":"2026-01-03T09:57:48.753Z","duration":20,"mimetype":"image/jpeg","is_enabled":true,"is_processing":true,"nocache":false,"play_order":0,"skip_asset_check":true}

4. See error
Response data: {
  error: "'<' not supported between instances of 'str' and 'datetime.datetime'"
}
Response status: 500
Response headers: {
  server: 'nginx',
  date: 'Fri, 03 Jan 2025 09:57:48 GMT',
  'content-type': 'application/json',
  'content-length': '80',
  connection: 'keep-alive',
  vary: 'Accept',
  allow: 'GET, POST, HEAD, OPTIONS',
  'x-frame-options': 'DENY',
  'x-content-type-options': 'nosniff',
  'referrer-policy': 'same-origin',
  'cross-origin-opener-policy': 'same-origin'
}

**Expected behavior**

this should return 200 successful

**Screenshots**

If applicable, add screenshots to help explain your problem.

[image]

**Environment (please complete the following information):**

- Device: [PC - anthias running in debian hyper visor]
- OS: [debian - latest]
- Anthias Version: v0.19.4
- Installation Method: [Manual Installation]

**Additional context**

no additional context

nicomiguelino, 2025-01-03:
@jay1459, thank you for bringing up this one.
In theory, `/api/v2/file_asset` should work without errors, as the web interface is currently using that endpoint.

Let me test and play around with the endpoint. If necessary, I'll apply fixes and/or update the API documentation on how to use `/api/v2/file_asset`.

In the meantime, feel free to try to set the content type to **_form-data_**.

nicomiguelino, 2025-01-03:
@jay1459, I just opened a pull request (#2191). One it gets merged, you can upgrade Anthias and see the fix take effect.

jay1459, 2025-01-03:
Thank you so much for the speed of this. Thats very appreciated.

nicomiguelino, 2025-01-03:
@jay1459, the fix was already deployed. Feel free to run the installer script again if you wish to update your Anthias instance.

nicomiguelino, 2025-01-03:
@jay1459, also feel free to re-open this issue if you still encounter the issue.

closed 2025-01-03 by nicomiguelino