Screenly/Anthias #2192: [BUG]

matze1708, 2025-01-04:
Hello,
I get a white screen on the monitor when I visit a certain page. 

On my display, we also have several assets that include images.

I have already tested several things, such as expanding the GPU memory.
The Pi is connected to a 55” TV via HDMI. 
But I have set the resolution to 1920x1080 in the cmdline.txt because otherwise the content was very small. 

I had found a similar problem when searching the net, which indicates a problem with the QTWebEngine. To do this, I had set the memory in docker-compose.yml.tmpl to 256MB, unfortunately without success, and in the Dockerfile for the viewer, I added: ENV QT_X11_NO_MITSHM=1, which probably also has something to do with the video memory, unfortunately also without success.
I had read that this is probably a general issue with QTWebEngine, but should be fixed with version 6.5.
If the URL is needed, I would like to send it via DM, not in the public area.

- Device: Raspberry Pi 4
- OS: Raspberry Pi OS 64-bit Bookworm
- Anthias Version: master@2d9db4b
- Installation Method: Installation Script 

Thank you very much.

nicomiguelino, 2025-01-06:
@matze1708, feel free to email the link(s) to nicomiguelino2014@gmail.com so that I can check them as well and sit on the issue.

matze1708, 2025-03-09:
Hello,
Some time ago, we had already exchanged the URL at Mail and had a little conversation about the problem.
Is there any new information on this or even an update that I can install on the PI4 that fixes the error?

Thanks a lot