Screenly/Anthias #2208: [BUG] Newest releases missing “docker-tag”

dougbot01, 2025-01-29:
I am following the manual installation instructions.

I used Raspberry Pi Imager to create a Raspberry Pi OS Lite image for my RasPi 5 board.

I am able to boot the image fine, using WiFi configured during the image process.

When running the installer, I have chosen to install a specific tag.
However, if I try v0.19.4 or v0.19.5 I get an error

“This version doesn’t have a docker-tag file.”

**To Reproduce**

Steps to reproduce the behavior:
1. From a Raspberry Pi OS Lite image, run $bash <(curl -sL https://install-anthias.srly.io)
2. choose tag instead of latest
3. Enter “v0.19.4” or “v0.19.5” for the tag
4. See error

**Expected behavior**

Installation should proceed

**Screenshots**

[image]

[image]

**Environment (please complete the following information):**

- Device: Raspberry Pi 5
- OS: Raspberry Pi OS Lite 64-bit Bookworm
- Anthias Version:  v0.19.5
- Installation Method: Manual Installation

**Additional context**

Add any other context about the problem here.

nicomiguelino, 2025-01-29:
@dougbot01, I just created `docker-tag` files for all Anthias releases that don't have that file yet. You could now re-run the installation script. Please let me know if you're still facing issues.

dougbot01, 2025-01-30:
Hi,

I was able to successfully install v0.19.5.

Thank you for the fast response and fix!

closed 2025-01-30 by dougbot01