Screenly/Anthias #2233: [BUG] Update cancelled with error Could not find a version that satisfies the requirement ansible-core==2.17.7

repher, 2025-03-15:
**Describe the bug**

Tried upgrade using:
`bash <(curl -sL https://install-anthias.srly.io)`

Updates canceledd with an error:

**Expected behavior**

Upgrade to current version.

**Screenshots**

[image]

**Environment (please complete the following information):**

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
$ cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.5

```

Installed via same command a few months ago an Raspberry Pi OS Lite.

Mimibs2025, 2025-03-18:
Same here with pi3b+ 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi 3 Model B Plus Rev 1.3

nicomiguelino, 2025-03-18:
@Mimibs2025, thank you for providing more details here.
I will look closer on this one. If all else fails, I could do a downgrade (of minor/patch version) of `ansible-core`.

Mimibs2025, 2025-03-19:
I did try to install Anthias on my pi3b+ several times (once with Debian bullseyes lite (32) , once with Debian bulleyes lite (64), still the same error : ERROR: Could not find a version that satisfies the requirement ansible-core==2.17.7
ERROR: No matching distribution found for ansible-core==2.17.7
The directory '/home/pi/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

closed 2025-03-25 by nicomiguelino

nicomiguelino, 2025-03-25:
@repher, @Mimibs2025 
I just  pushed the fixes.
Please try running `bash <(curl -sL https://install-anthias.srly.io)` again and let me know if you're still having the same issue.

repher, 2025-04-15:
Hello!

Maybe unrelated but the following error happened.

Error on first run of `bash <(curl -sL https://install-anthias.srly.io)` with all options selected by default:
[image]

Then:
```
sudo apt-get clean
sudo apt-get update
sudo apt-get upgrade
sudo apt-get autoclean
```

Second try `bash <(curl -sL https://install-anthias.srly.io)`:
update worked

This was was also reproducible on a second device.

Thanks!
Bernhard

PR #2248 (2025-03-25): fix: installing Anthias in Bullseye
### Issues Fixed

- Fixes #2233

### Description

- Reverts `ansible-core` from `2.17.7` to `2.15.9` for Bullseye and older
- Bumps `ansible-core` to `2.18.3` for Bookworm

### Checklist

- [x] I have performed a self-review of my own code.
- [x] New and existing unit tests pass locally and on CI with my changes.
- [x] I have done an end-to-end test for Raspberry Pi devices.
- [x] I have tested my changes for x86 devices.
- [x] I added a documentation for the changes I have made (when necessary).